morrownr/8812au #1: Debian 10.6 (kernel 5.9.0-1-amd64)

developerofshadows, 2020-11-15:
Hello, i can't build your drive and get next error, when type make in console: 
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.9.0-1-amd64/build M=/home/work/gits/8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-5.9.0-1-amd64'

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

make[2]: *** [/usr/src/linux-headers-5.9.0-1-common/Makefile:718: include/config/auto.conf] Error 1
make[1]: *** [/usr/src/linux-headers-5.9.0-1-common/Makefile:185: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.9.0-1-amd64'
make: *** [Makefile:2395: modules] Error 2

morrownr, 2020-11-15:
I currently don't test with Debian due to a lack of time to accomplish everything that is on my to-do list but maybe that will change in time.

I will try to help: my first thought is that you don't have the appropriate kernel headers installed. The provided instructions for Mint/Ubuntu probably don't help with that for Debian. Here is a link to an article that should help you.

There may be additional things that need to be installed in Debian but first lets see if installing the proper headers works.

Let me know and if we get this sorted out I'll add notes for Debian users to the REAME. Good luck.

developerofshadows, 2020-11-16:
Hello again! Sorry for my last message, yesterday , after open this issue, i tried to reinstall linux-headers, but debian freeze on last step (when processing triggers). And this give me the idea, that the problem with this package! To confirm my theory i began to reinstall the whole system, and it's worked! Your driver is perfectly worked on bullseye (testing) version of debian.

closed 2020-11-16 by developerofshadows