morrownr/8812au #2: Original Realtek 5.9.3.2 driver

gordboy, 2020-11-26:
Hello

I would be very interested to be able to get the original Realtek 5.9.3.2 driver, as the last one I was able to get from FTP was 5.6.4.2. The FTP credentials I have are no longer working.

My repo is starting to show signs of wear and tear, and it would be good to rebase to a newer version.

My 5.6.4.2 repo

https://github.com/gordboy/rtl8812au-5.6.4.2

morrownr, 2020-11-26:
gordboy, I'll be glad to drop off a clean copy. Let me catch up on some things and I'll get it over to your place as a zipfile.

Nick

gordboy, 2020-11-26:
Thanks I grabbed some stuff from your commit at

https://github.com/morrownr/8812au/tree/9f1cafe7aefa34a48ca21c8da94ca84d8de25749

and ditched all the top level files except Makefile and Kconfig. But a pristine clean copy complete with changelog and stuff would be nice.

closed 2020-11-26 by gordboy