morrownr/8812au #4: Injection not functional on 5ghz

Markw27348, 2020-12-01:
Injection fully working on 2.4ghz, 5 ghz it is not effective at all.

morrownr, 2020-12-01:
Markw, thanks for the report. I'll start looking at this. Can i get you to make me a checklist that reflects the steps you are taking, the results and what the expected results are? That would help me greatly.

FYI: I have little experience pen testing but will do my best to make that capability work.

morrownr, 2020-12-02:
I had time to test this evening. Here is my initial test using this driver on a Raspberry Pi 4B using the 32 bit Raspberry Pi OS with Aircrack:
 
$ sudo aireplay-ng --test wlan0
00:03:44  Trying broadcast probe requests...
00:03:44  Injection is working!
00:03:46  Found 4 APs

00:03:46  Trying directed probe requests...
00:03:46  48:77:46:D6:D7:E7 - channel: 149 - 'myAP'
00:03:46  Ping (min/avg/max): 1.468ms/6.256ms/47.965ms Power: -48.80
00:03:46  30/30: 100%

00:03:46  42:77:46:D6:D7:E7 - channel: 149 - ''
00:03:46  Ping (min/avg/max): 0.476ms/4.938ms/10.000ms Power: -47.50
00:03:46  30/30: 100%

00:03:46  46:77:46:D6:D7:E7 - channel: 149 - ''
00:03:46  Ping (min/avg/max): 2.397ms/5.070ms/7.712ms Power: -46.27
00:03:46  30/30: 100%

00:03:46  12:59:32:04:A7:B8 - channel: 149 - ''
00:03:46  Ping (min/avg/max): 0.606ms/3.004ms/6.278ms Power: -42.77
00:03:46  30/30: 100%

Channel 149 is definitely 5g. I need more information.

Markw27348, 2020-12-02:
Aireplay-ng -test command works, but a simple deauth using aireplay-ng or mdk3/mdk4 does not

Zanzaux, 2020-12-08:
> Aireplay-ng -test command works, but a simple deauth using aireplay-ng or mdk3/mdk4 does not

Hi are you trying an 802.11ac access point?

morrownr, 2020-12-08:
Markw27348,

I am not ignoring you but you have not providing the information that I requested:

"Can i get you to make me a checklist that reflects the steps you are taking, the results and what the expected results are? "

I am not a pen tester and have no experience along those lines. I am working to make this driver as fully functional as possible but I have limited amounts of time to spend on each issue. Think of it like this: Unless i can recreate the problem, I am unlikely to be able to fix it. Think about what I would need to be able to recreate the problem.

closed 2020-12-14 by morrownr