morrownr/8812au #5: Still no support for Virtual Interface

usama7628674, 2020-12-03:
Just tried this driver,looks good and dmesg log is clean,but still no support for virtual interface.I couldn't do injection and AP at the same time.Seems like realtek won't write driver for `mac80211` ever.You said on `gordboy` repo that you've code for virtual interface and I want to ask you is this really the case?

morrownr, 2020-12-03:
Well, actually I think I said that I was waiting for gordboy to submit the code so I can commit it. I also put a <g> in an attempt to suggest that I was joking. I do not have any code supporting virtual interface. I am capable of adding some capabilities such as LED control. I am also capable of of maintaining the driver as kernel changes are made but virtual interface would be difficult. That is likely something that only Realtek can do as they are the only ones that have full knowledge of all of the details.

lokpimkd, 2020-12-03:
> Just tried this driver,looks good and dmesg log is clean,but still no support for virtual interface.I couldn't do injection and AP at the same time.Seems like realtek won't write driver for [...]

Buddy instead of begging here and on godrboy's repo maybe its better for now to use dual adapters. You can buy any cheap adapters that is capable of creating the AP for around 20$, and use the 8812au for deauth and packet injection.

usama7628674, 2020-12-04:
Lol, I wasn't begging I got amped when I read your comment on gordboy repo.Nevermind, I have three adapters already.

closed 2020-12-04 by usama7628674