morrownr/8812au #6: Not working Kali 2020.4

Zanzaux, 2020-12-08:


morrownr, 2020-12-08:
I will assume you installed this driver because you have a USB WiFi adapter based on the Realtek 8812au chipset? Please confirm.

I am not currently a user of Kali but I think I have read where Kali comes with a driver for adapters based on the 8812au chipset. If that is the case, you will need to uninstall or blacklist that driver or there may be problems.

Zanzaux, 2020-12-08:
Yes, I have an ALFA awus036ach RTL8812AU, until a few days ago I used another driver with DKMS installation today I reinstalled kali but I was unable to test your driver, I'm looking for a driver that also enables "special" frequencies like 5460 5440 5030 5040 5060 unfortunately I can't detect them using airodump-ng wlan1 -C

morrownr, 2020-12-09:
Zanzaux, right now I am very busy doing things like working on the updates for kernel 5.10. I also have a relatively small lab which is why I am limiting testing to RasPiOS, Linux Mint Debian Edition, Linux Mint and Ubuntu. I enjoy doing this... keep in mind I am maintaining 5 separate drivers on this site,,, but I cannot be everything to everybody. This is an issue I simply do not have time to investigate. This is, as far as I can tell, the newest code base of this driver that is available and it is a solid code base... far better than previous editions. If you can investigate and make suggestions, I will be glad to hear from you.

Nick

closed 2020-12-09 by morrownr