morrownr/8812au #7: Info: Running **hostapd**, Access Point, with this driver...

morrownr, 2020-12-10:
I am currently testing this driver with hostapd. With the proper driver settings and hostapd.conf settings, I have found that this driver and a good rtl8812au adapter make for a very good Access Point. All capabilities, including n and ac are supported, including channels widths up to 80.

I plan to make a guide available here soon. If anyone wants to start testing now, let me know and I will post a short version of the guide here. I will eventually make the details available in the main documentation.

Nick

usama7628674, 2020-12-10:
Alfa AWUS036ACH is an excellent adapter with a built-in amplifier as well.

morrownr, 2020-12-10:
The Alfa AWUS036ACH adapter is outstanding. I have one. It can be a better AP than many, if not most, dedicated WiFi routers. This is a case where paying more can be worth it. And the driver here is probably the best driver I have ever seen Realtek kick out. I only had to clean one minor issue in the code and the rest of my time has been spent working on enhancements and documentation. This driver works and works well. I just finished a multi-day test with this adapter and hostapd and wow. Great results.

closed 2021-01-03 by morrownr