morrownr/8812au #9: Unable to change MAC address

oldstanda, 2021-01-07:
I have Ubuntu 18.04 with the 5.4.0.59 kernel and in the current version 5.9.3.2 it is not possible to change the MAC address: RTNETLINK answers: Operation not permitted.
Version 5.9.3.2 from November 2020 works.

morrownr, 2021-01-07:
Can I get you to give me a step by step list of what you are doing so I can try to duplicate it.

Nick

oldstanda, 2021-01-07:
I use a simple bash script: 

```
airmon-ng check kill
ip link set wlan2 down
iw wlan2 set monitor control
ip link set wlan2 address 00:11:22:33:44:55 up
```

morrownr, 2021-01-08:
Okay. It may take a couple of days but this is on my list so I will get to it.

morrownr, 2021-01-11:
I had a chance to look at this issue today. The test system I used is an amd64 system running Ubuntu 20.10. I used your script and was able to duplicate your results. At that point I did some searching on the internet. I found a lot of information. I wasn't able to reach a definitive conclusion but came away thinking that it is unlikely this driver is the cause. If you can research the issue and provide specific information that shows the driver is causing this problem, I'll be glad to take a look. Please understand that these drivers are incredibly complex and we do not have full access to all of the source code as Realtek does not provide everything in human readable form. Regards.

oldstanda, 2021-02-06:
https://youtu.be/XpkaUlBB6AQ

morrownr, 2021-02-08:
@oldstanda 

I'm finally getting time to configure a test system with a 8812au based adapter.

Before I dig into this issue, can I get you to try this driver:

https://github.com/gordboy/rtl8812au-5.9.3.2

Gordboy and I started with the same source code. If I can get you to try his edition of this version of the driver, that could go a long way toward helping me determine what the problem is as I can look at the differences in the source of the two drivers.

oldstanda, 2021-02-08:
With the driver from gordboy changing the mac address works. Deauth does not work, which did not work with previous versions of the driver.

morrownr, 2021-02-08:
Well, that is good information. I have tools that can rapidly compare Gordboy's source to my source and show me the differential.

It is starting to look like something I did is indeed causing the problem. Let me see what I can figure out.

morrownr, 2021-02-08:
Try this:

Change MAC Address and Enter Monitor Mode

$ sudo airmon-ng check kill

$ sudo ip link show

$ sudo ip link set dev <your device here> down

$ sudo ip link set dev <your device here> address <your new mac address>

$ sudo iw <your device here> set monitor control

$ sudo ip link set dev <your device here> up

$ iw dev

-----
I did a lot of work going over the source but could not find anything that I thought would contribute to the issue you are seeing. The work is okay because I needed to do some house cleaning anyway. Now that I have worked on the source, I could really use some help testing. Can I get you to run the ./remove-driver.sh script followed by deleting the driver directory and then do a clean installation? It would really help.

I added the above to the document called "Monitor Mode using iw and ip.md." Would you be interested in working on that document as it could certainly be improved and become more useful to others?

Let me know if the above gets you where you are trying to go. It certainly seemed to work here.

oldstanda, 2021-02-09:
Super! 
You divided the address + up separately. It didn't occur to me ;(
Thank you very much