morrownr/8812au #12: Kali 2020.4 USB3.0 and High power does not work！

LuckyFishGeek, 2021-01-13:
kali kernel:Linux kali 5.9.0-kali1-amd64 #1 SMP Debian 5.9.1-1kali2 (2020-10-29) x86_64 GNU/Linux

set usb3.0
[image]

The result after restarting the virtual machine
[image]

To set high power, I used the following command

sudo iw reg set US

sudo ip link set wlan0 down

iw wlan0 set txpower fixed 3000  

sudo ip link set wlan0 up

Return result

[image]

I don't understand where is the problem? Can you help me?

thanks

morrownr, 2021-01-13:
How about we get to the bottom of the USB 3 problem, then we can take on the power problem.

I have a relatively small computer lab with limited test systems so I can't keep too many distros available for testing at any time. I did load Kali temporarily for testing a few months ago and it memory serves me correctly, a driver for the rtl8812au ships with Kali. I don't remember how it was included so I don't know if it needs to be blacklisted or if it is even a problem. You might want to investigate if you haven't already.

Are you using a cable? If so, is the cable a USB 3 cable?

Is the adapter/cable in a USB 3 port? If it is a desktop, it is a USB 3 port on the back of the system?

Is USB 3 turned on in BIOS?

I'm asking the questions because USB 3 support in this driver is very dependable and you appear to have it turned on properly... so why is it not working? It is not clear to me.

LuckyFishGeek, 2021-01-13:
> How about we get to the bottom of the USB 3 problem, then we can take on the power problem.
> 
> I have a relatively small computer lab with limited test systems so I can't keep too many distros available for testing at any time. I did load Kali temporarily for testing a few months ago and it memory serves me correctly, a driver for the rtl8812au ships with Kali. I don't remember how it was included so I don't know if it needs to be blacklisted or if it is even a problem. You might want to investigate if you haven't already.
> 
> Are you using a cable? If so, is the cable a USB 3 cable?
> 
> Is the adapter/cable in a USB 3 port? If it is a desktop, it is a USB 3 port on the back of the system?
> 
> Is USB 3 turned on in BIOS?
> 
> I'm asking the questions because USB 3 support in this driver is very dependable and you appear to have it turned on properly... so why is it not working? It is not clear to me.
> 
> Oh, you did not mention if this system is amd64 or a RasPi (arm)...?

Thank you for your help. I am a novice, so I may not express well when asking questions. Please forgive me!

My system is amd64

The USB3.0 interface and data cable are used

USB3.0 settings are also turned on in the BIOS

Thank you again for the optimized driver. It runs very well!

morrownr, 2021-01-13:
You are doing fine. It can take a little time to be fully comfortable with a new OS. We were all novices at some point. For me it was in the mid-90's. Back then we didn't have distros. e had to figure out how to get the kernel on the system and then... it was fun.

Let me continue with some questions because not seeing USB 3 working tells me something is not right.

Can I get you to plug the adapter directly into a USB 3 port? It doesn't matter if it is not in a place to get a good signal, this is for testing only. See if you see 5000m. If you do, there is a problem with the cable. If you do not, keep retesting in each port that is supposed to be USB 3 until you either see 5000 or don't after all are tested.

Exactly what adapter do you have?

closed 2021-01-13 by LuckyFishGeek