morrownr/8812au #13: 802.1x dynamic VLANs

prbluebottle, 2021-01-30:
Hi,
Are dynamic VLANs supported by this driver? It's difficult to find out because not many people use that feature.
Thanks

morrownr, 2021-01-30:
I am not familiar with this capability. Can you provide a link so that I can educate myself on this topic.

Regards.

prbluebottle, 2021-01-31:
OpenWRT can be used for this, if the driver supports AP/VLAN mode: https://openwrt.org/docs/guide-user/network/wifi/wireless.security.8021x#x_dynamic_vlans_on_an_openwrt_router
Thanks.

morrownr, 2021-02-01:
Okay, I have a little better understanding now. I still don't have experience with it so take what I say as a best guess based on a little research. I concentrate my efforts on client and ap modes.

Of the 80211AC capable adapters that I have, the only one that is capable of AP/VLAN on Linux is the adapter based on the mt7612u chipset. None of the adapters based on Realtek chipsets are capable due to driver limitations. This is not a capability that is likely to happen out here in the user community. Realtek would need to add this capability.

Regards

prbluebottle, 2021-02-02:
Yep, that's what I thought the situation might be. Thanks for taking a look.

morrownr, 2021-02-06:
prbluebottle,

I was doing some work this morning with an adapter that uses the mt7612u chipset. Listed capabilities:

	software interface modes:
		 * AP/VLAN
		 * monitor
	valid interface combinations:
		 * #{ IBSS } <= 1, #{ managed, AP, mesh point, P2P-client, P2P-GO } <= 2,
		   total <= 2, #channels <= 1, STA/AP BI must match

That caught my eye so I thought I would pass this on to you. I'm not sure if that is what you are looking for. You can find links to adapters that use the mt7612u chipset here:

https://github.com/morrownr/USB-WiFi

FYI: I am currently testing an adapter with the 7612u chipset on a RasPi4b with RasPi OS (32 bit)(kernel 5.10). The driver for this chipset is in-kernel and so far seems very solid. I have confirmed WPA-3 SAE support so far. I can post additional test results when available if you are interested.

prbluebottle, 2021-02-08:
Thanks for the info, might end up buying one of those.

closed 2021-02-08 by prbluebottle