morrownr/8812au #14: Aircrack-Ng segmentation fault and crash

DeityOfChaos, 2021-02-19:
Manual Monitor mode using iw will work, but tools that automatically set the monitor mode will crash, like bettercap tool or "sudo airmon-ng start wlan1"

This git is by far the best driver I found, fixed deauths  - thank-you

morrownr, 2021-02-19:
You are welcome. I started this repo about 4 months ago with freshly available source code. I noticed immediately that this driver is of seriously better quality than anything I have ever seen from Realtek and it is of vastly better quality than previous 8812au releases from Realtek. Of the 5 drivers I maintain ( github.com/morrownr ), this is by far the best driver. It is outstanding in AP mode. It is the only Realtek driver that is fully stable using 80 MHz channel width in AP mode.

I am aware of some of the issues with this and other Realtek drivers and airmon-ng. I took the time recently to look at the airmon-ng source. I backed away. It takes almost all of the spare time I have to maintain the 5 drivers and informational sites I work on. I welcome good coders and testers to help me. so maybe I have extra time. Maybe you can write your own scripts to automate what you are doing? I might even find the time to help and we could include copies with this driver.

DeityOfChaos, 2021-02-20:
I am glad you confirm the AP mode works properly, I had projects with that in mind. Yes, I will be using scripts that bypass this problem automatically, which should be sufficient for airmon-ng (hopefully). This is a fantastic git though, I am grateful :)

closed 2021-03-11 by DeityOfChaos