morrownr/8812au #15: Dunno what you did

DeityOfChaos, 2021-03-03:
sometime this week, yesterday I believe, you must have changed something that makes my device get 42949572.96dBm with iw dev on managed mode. Fresh install from this branch and it goes like that (I use your repo a lot so I noticed it)

anyone else?

morrownr, 2021-03-03:
I checked the log. The only recent code change was yesterday and it had to do with an update to support k5.12. It had nothing to do with TX values. Everything else that has been changed in the last week has been changes to documentation.

Do a reboot and tell me if you see the problem. If you do, I need the steps you are taking in order to duplicate the problem.

With that said, let me wax poetic for moment. These Realtek drivers don't fully support iw and ip. Some of what they do support gives bad info. The power saving info... oh, don't trust that for a moment. If you want a USB WiFi adapter to use with Linux that fully supports iw and ip, go here: https://github.com/morrownr/USB-WiFi

I recommend adapters based on the mt7612u chipset. The driver is really good and is in the kernel. This means that it is compliant with Linux wireless guidance. It also means that you can file a bug report with reasonable expectation that the issue will be worked. Have you tried to file a bug report with Realtek lately? Good luck with that.

If it isn't obvious by now, I work on the drivers on this site as a way to give back to the community. That does not mean that I support the way Realtek supports Linux nor do I recommend their products if someone asks. I have and use adapters with Realtek chipsets and Mediatek chipsets. I am able to compare the products. My opinion is that Linux users are better served with the Mediatek based adapters. 

Probably more than you wanted to know. Have a good evening.

DeityOfChaos, 2021-03-03:
Thanks. Unfortunately I already have an order of 3 different wifi adapters incoming lol should have checked your list first. thankfully one of them is AWUS036NHA which I've read is well compatible.
You recommend the Alfa AWUS036ACM too bad I bought the AC version, which uses this driver. the only reason I chose one over another was the higher output power of AC over ACM x')

DeityOfChaos, 2021-03-03:
The steps I followed were:
Install fresh Raspbian 32 bit with Raspberry Pi imager
Turn on device, apt-get update/upgrade etc

git clone https://github.com/morrownr/8812au
cd 88*
raspi32.sh
install-driver.sh

iw dev 
>absurd dBm 

(never had seen it happening with this repo, switched back for aircrack-ng's maintained driver and it went back to the original and standard 20dBm value)

DeityOfChaos, 2021-03-03:
Another question, where and how often does Realtek release code for 8812au? From what I've seen, most 8812bu ones are relatively low power which sacrifices range. Can you check how much mA output power you get on Cudy's or any 8812bu driver? Not sure if lsusb -v output is reliable but this is what I get:

My Alfa AWUS036AC:

[image]

morrownr, 2021-03-03:
I'll gather the requested information plus some test data on a sample of adapters, including the Cudy you requested, and post that back here, hopefully today.

Keep in mind that you can return products.

morrownr, 2021-03-04:
Quote: "Another question, where and how often does Realtek release code for 8812au? I can't find official source code, download or linux related stuff at all."

Now you are starting to understand why I try to push Linux users to Mediatek based adapters.

The answer to your question: Where? Nowhere. Realtek does not make drivers available to consumers. How often? I am not aware of a schedule and Realtek certainly doesn't provide one.

You will have a hard time finding official source code. I have to scan select sites where a select number of adapter makers post, at times, an updated source they likely got from Realtek. Are you sure you want a clean copy of the source? I can give you a clean copy of the latest I have for each driver but it may not go as smooth as you think. Unmaintained Linux driver source code dies from bit rot fairly quickly. I put a lot of work into keeping these drivers going.

A question you did not ask but is important is: Where at Realtek do I report bugs? Thanks for asking. I needed a good laugh. There is no official channel to report bugs to Realtek from a consumer. I'd love to report some bugs but I can't... that is why I push Linux users to Mediatek based adapters.

morrownr, 2021-03-04:
2012-03-03

AC1200+ USB 3 WiFi Adapter Comparison

Adapter					Chipset					USB Capability
MaxPower				Device ID				Visible Antennas
iperf3 Test				Link Quality			Signal Level

(1)
Alfa AWUS036ACH         rtl8812au               USB3
800mA					ID 0bda:8812			2
309 Mb/s				Link Quality=99/100  	Signal level=-43 dBm

(2)
Cudy WU1400             rtl8812bu               USB3
504mA                   ID 0bda:b812            1
291 Mb/s                Link Quality=84/100     Signal level=-51 dBm     

(3)
Alfa AWUS036ACM         mt7612u                 USB3
400mA                   ID 0e8d:7612            2
237 Mb/s                Link Quality=79/100     Signal level=-55 dBm

(4)
Generic (brown box)     rtl8812bu               USB3
504mA                   ID 0bda:b812            2      
216 Mb/s                Link Quality=70/100     Signal level=-61 dBm

(5)
Generic (brown box)     rtl8814au               USB3
864mA                   ID 0bda:8813            2
204 Mb/s                Link Quality=86/100     Signal level=-50 dBm

Test conditions:

Distance from adapters to AP was about 40 feet (2 walls).

iperf3 server is running on a Raspberry Pi 4b that is connected to the
network via CAT 6 ethernet cable.

Adapters were connected to a rear USB 3 port on a Dell Optiplex 9020
with an i7 processor running Linux Mint 20.1.

Utilities used include wavemon, iw and iperf3.

iperf3 test: $ iperf3 -c 192.168.1.40 -t 60

Analysis:

This is a limited test based on a user request. I'll plan on expanding
the test as time permits.

If I have to declare an overall winner from the current top 3 adapters,
it would be this adapter:

Alfa AWUS036ACM

The reason for picking the above adapter is that it did a good job in
all categories.

Here are the categories.

-----

If price is important:

Cudy WU1400			(low price)
Alfa AWUS036ACM		(mid price)
Alfa AWUS036ACH		(very high price)

-----

If a strong signal is important:

Alfa AWUS036ACH		(astoundingly good)
Cudy WU1400			(very good)
Alfa AWUS036ACM		(very good)

-----

If having the adapter supported in the Linux kernel is
important, so that it is not necessary to find, download,
and compile a driver, of the ones tested, this is your only
option:

Alfa AWUS036ACM		(it is a very good driver)

-----

If AP mode is important:

Alfa AWUS036ACM
Alfa AWUS036ACH

-----

If fast managed (client) mode is important:

Cudy WU1400
Alfa AWUS036ACH
Alfa AWUS036ACM

-----

If monitor mode is important:

Alfa AWUS036ACM
Alfa AWUS036ACH

-----

If low power usage is important, such as if you are
going to use it with a Raspberry Pi 4b:

Alfa AWUS036ACM
Cudy WU1400

-----

If not having to worry about future driver support
is important:

Alfa AWUS036ACM
Cudy WU1400

-----

If having the the device driver support current Linux
Wireless standards is important:

Alfa AWUS036ACM

-----

If having a clear path to report driver problems is
important:

Alfa AWUS036ACM

-----

Link to MT76 driver in the Linux kernel:

https://github.com/torvalds/linux/tree/master/drivers/net/wireless/mediatek/mt76

-----

Link to RTL88X2B driver in the Linux kernel:

https://github.com/torvalds/linux/tree/master/drivers/net/wireless/realtek/rtw88

Note: This driver does not support usb adapters yet but there is hope.

-----

Link to RTL8812AU driver in the Linux kernel:

It does not exist and likely will never exist. My opinion is that we will never
see a fully Linux Wireless standards compliant driver for the rtl8812au chipset.

-----

Hopefully this document is of help to Linux users that want or need USB 3 WiFi
adapter support. I will look to expand this document in the future.

Nick

DeityOfChaos, 2021-03-04:
> I checked the log. The only recent code change was yesterday and it had to do with an update to support k5.12. It had nothing to do with TX values. Everything else that has been changed in the last [...]

Oh gosh I didn't realize they were that useless, what the hell is this kind of service? why do manufacturers even mass produce wireless cards knowing realtek does nothing? oof

Thanks for the data, hopefully people around the web will find this useful.
I can't find where on the web but I've seen somewhere that Alfa advertises AWUS036AC as "high power" and AWUS036ACM something like medium power or high power, whatever inferior degree of power in their scale: it seems that the difference is just from 500mA to 400mA lol I wish I could cancel my Alfa Network order cus it's all Realtek chipsets but it should be arriving tomorrow

DeityOfChaos, 2021-03-04:
Refreshed their wiki 

[image]

morrownr, 2021-03-04:
Go get 'um. I think the thing that will get Realtek's attention is when they see a shift in product sales that is costing them money. The information from ( https://github.com/morrownr/USB-WiFi ) should be spread far and wide. Linux users do have a choice and we can vote with our money.

DeityOfChaos, 2021-03-11:
Just bought Cudy WU1400 from Amazon. I emailed them asking if they had new official drivers from realtek for 8812AU they said no, but they sent me v5.8.5.1 for 8814au. They absolutely rock. Very good support honestly, I bought it to support them. I'll test the usb device in AP mode, it's cheap and I'll be making a 2.4Ghz wifi signal repeater with a raspberry pi. (Of course I could use AC/ACM/ACH but they are too expensive and I am borrowing the pi to a friend so god knows what will happen)

closed 2021-03-11 by DeityOfChaos