morrownr/8812au #16: Problem with kernel 5.4.83-v7l+ on RPi

pavetrck, 2021-03-09:
Hi, after whole process i got this error message...

DKMS: add completed.
Error! echo
Your kernel headers for kernel 5.4.83-v7l+ cannot be found at
/lib/modules/5.4.83-v7l+/build or /lib/modules/5.4.83-v7l+/source.
An error occurred while running: dkms build : 1

------------------------------
Deleting module version: 5.8.5.1
completely from the DKMS tree.
------------------------------
Done.
Removing 8814au.conf from: /etc/modprobe.d
Removing source files from: /usr/src/rtl8814au-5.8.5.1
Cleanup complete

... so the conf file missing

What Iam I doing wrong? I run on 2021-01-11-raspios-buster-armhf-lite, RPi 4B

Thx

morrownr, 2021-03-09:
You may not be doing anything wrong. For me to attempt to duplicate the problem I need some more information.

This issue is posted in the repo for the 8812au driver but your posting seems to indicate you were installing the 8814au driver. We need to clarify this issue, Can you tell me exactly what USB WiFi adapter you are using?

Also, I see that you are using a rpi4b. Can you tell me what else is plugged into the pi? Do you have powered hub or is the adapter plugged directly into a port on the pi?

pavetrck, 2021-03-09:
Thanks for your quick reply. I use ASUS Dual band AC1900 USB 3.0 Wi-Fi Adapter (USB-AC68). Adaper is plugged directly in RPi. In RPi is plugged moreover only keybord and i use eth0 interface as well.

morrownr, 2021-03-09:
Nice adapter. It is based on the 8814au chipset so it appears you have the correct driver. Just be aware that you are posting the issue in the 8812au repo.

Before we get started, I need to make sure you are aware that the rpi4b is limited to a total of about 1200 mA in the usb subsystem and the 8814au can take most of that when pushed. I have metered my 8814au to around 850 mA when pushed so you don't have much extra power to play with. It works, just don't start plugging extra things like an SSD directly into the pi also or it is brownout time.

My rpi4b is running the 5.10.17 kernel. I thought the January release of the OS came with 5.10.17. So, how is it that you are running the 5.4.83 kernel? I'm sure we can figure it out if there is a need but how about you run these 3 lines and try the install again...

$ sudo apt update
$ sudo apt full-upgrade
$ sudo apt-get install -y raspberrypi-kernel-headers bc build-essential dkms git

Make sure you run the following line per the instructions:

$ ./raspi32.sh

All indications are that you are running the 32 bit release version so my advice is based on that. If you are running the 64 bit beta version, we'll need to start over.

Let me know.

pavetrck, 2021-03-10:
Hi, its my stupid fault... I did not upgrade system (only update). Now its working fine. Only one problem...  same with 8814au and 8812au... After switching adapter to monitor mode and starting scan (airodump) only clients are visible first of all and after long time accsess points are visible. And in comparsion with other adapters these two see less APs...

morrownr, 2021-03-10:
I'll investigate. I need you to to write a step by step list of what you are doing so that I can try to duplicate what you are seeing.

I have a rpi4b and adapters based on the same two chipsets you are using so I should be able to duplicate what you are seeing.

Thanks.

pavetrck, 2021-03-11:
HW: RPi 4B 4GB RAM, only keyboard in USB port, adapters ASUS Dual band AC1900 USB 3.0 Wi-Fi Adapter (USB-AC68) and ALFA AWUS036ACH (AC1200)
SW: Clear image of Raspberry Pi OS Lite 2021-01-11 lite, apt update, apt full-upgrade, your drivers 8812, 8814, reboot, apt install aircrack-ng, apt install wavemon

Everything works fine in managed mode of both adapters (tested in wavemon)

But after switching to monitor mode, problem with scaning AP as I mentioned.
Here is screen of iwconfig command after switching to monitor mode (trying with airmon-ng start wlan1 or iwconfig wlan1 mode monitor - same result)

wlan1     IEEE 802.11b  ESSID:""  Nickname:"WIFI@RTL8812AU"
          Mode:Monitor  Frequency:2.412 GHz  Access Point: Not-Associated
          Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

Scanning (airodump-ng start wlan1)... much more APs are physicaly present

 CH  5 ][ Elapsed: 6 s ][ 2021-03-11 09:18

 BSSID              PWR  Beacons    #Data, #/s  CH  MB   ENC  CIPHER AUTH ESSID

 DA:47:32:60:72:A2  -45       94        0    0   1   54  WPA2 CCMP   PSK  <length:  0>
 D8:47:32:40:72:A2  -45       92        0    0   1   54  WPA2 CCMP   PSK  TP-Link_72A2
 4A:89:E7:55:6C:D4  -78       77        3    0   1   54  WPA2 CCMP   PSK  tartan

 BSSID              STATION            PWR   Rate    Lost    Frames  Probe

 (not associated)   44:85:00:31:90:AC  -71    0 - 1      0        1
 (not associated)   2E:C0:4A:E9:81:C2  -55    0 - 1      1        3
 (not associated)   D8:C0:A6:BF:EB:33  -53    0 - 1     11        6
 (not associated)   44:85:00:31:90:43  -73    0 - 1      8        5  
 (not associated)   9A:96:9C:35:E9:60  -77    0 - 1      0        1
 (not associated)   E2:4A:5D:3E:FC:37  -77    0 - 1      1        2
 (not associated)   DE:91:B3:6C:2D:0E  -77    0 - 1      0        1
 (not associated)   B2:B4:E9:37:96:F0  -79    0 - 5      0        1  
 4A:89:E7:55:6C:D4  6E:AA:E4:2F:7E:75  -23    0 - 1      0        5
 4A:89:E7:55:6C:D4  8C:B8:4A:81:CA:A3  -81    0 - 1      0        3

DeityOfChaos, 2021-03-11:
> HW: RPi 4B 4GB RAM, only keyboard in USB port, adapters ASUS Dual band AC1900 USB 3.0 Wi-Fi Adapter (USB-AC68) and ALFA AWUS036ACH (AC1200)
> SW: Clear image of Raspberry Pi OS Lite 2021-01-11 lite, [...]

I am also using RPI for these kind of projects, I own AWUS036ACH,ACH,NHA and Soon, Cudy WU1400:

The 8812AU driver is working fine for AWUS036ACH on Pi4. I detect pretty much all APs in a huge radius.

Try to reboot pi and use iw commands instead of iwconfig:

sudo ip link set wlan1 down
sudo iw dev wlan1 set type monitor
sudo ip link set wlan1 up

(for channel: 
sudo iw dev wlan1 set channel 1)

Try:
airodump-ng interface 
airodump-ng -c(Channel) interface (will only scan in one channel)
airodump-ng -b abg interface
and then again, and see if anything changed in output.
airodump-ng interface

If it persists I'd recommend swapping to Aircrack-ng's drivers: https://github.com/aircrack-ng/rtl8812au/issues/794
and see if problem persists. Let me tell you I prefer morrownr's driver 1000x over Aircrack-Ng's driver, but sometimes bugs can be found in one and gone in the other. I believe aircrack-ng uses older versions of the driver and different bugs are patched.

Let me warn you that sometimes bugs come and disappear out of nowhere. I've had problems like this, I've also had channel hopping issues with 'airodump-ng -c11 wlan1' (it was only supposed to get channel 11 but got 8 or 10 too etc). Try different commands 

I've made my own python3 script using Scapy to scan for APs / stations pretty much like aircrack-ng using Scapy and it found just as many APs as aircrack-ng without any channel-hopping issues, so honestly sometimes the bugs can be from aircrack itself too.

morrownr, 2021-03-11:
@DeityOfChaos 

Quote "I am also using RPI for these kind of projects, I own AWUS036ACH,AC (both RTL8812AU) and AWUS036NHA (atheros 9k). Soon, Cudy WU1400 which is RTL8812BU which seems to be replacing RTL8812AU in TP-Link and many other manufacturer's models"

I have a Cudy WU1400. It is a pretty good quality adapter but if given the choice between it and my Alfa AWUS036ACM, I will take the Alfa AWUS036ACM every time. Why? It uses a Mediatek mt7612u chipset. The driver is really good and is in the kernel which means we can watch it and even submit bug reports. It supports WPA3 and other things that these Realtek drivers do not. My advice is to not spend new money on Realtek based adapters.  More into:

 https://github.com/morrownr/USB-WiFi

morrownr, 2021-03-11:
@pavetrck 

I'm getting a little behind right now so it may take a while for me to look at this. Yell at me again if you think I forgot you.

Nick

DeityOfChaos, 2021-03-12:
> re into:

I didn't buy ACM one because where I live, instead of 15-20$ that I spent on Cudy, I would be paying 40-50€++ which is probably around 60$. Its a model with low availability around here. I bought cudy for several reasons: Cheap, I am supporting them, won't bother me if it breaks because I'll borrow it ;)

morrownr, 2021-03-15:
@pavetrck 

I had a little time to look at your issue this evening. I have noticed this myself but have not been able to track down the cause. Something I have noticed is that the source we have for the 8814au is somewhat dated and when you compare it to the code for the 8812au, wow, there is a BIG difference. Many things work well in the 8812au driver so I have used meld to see if things that may have been fixed or updated can be moved to the 8814au driver but the code in the 8814au just gives me the impression that it is old and stuck in time. What I have done is limited my 8814au adapter to managed mode duty and nothing else. What we really need is a new version of the 8814au driver from Realtek.

The best adapters for master mode and monitor mode, as far as what I own is concerned, are my adapters based on the my7612u and 8812au chipsets.

A little behind the scenes information: First and foremost I am a user just like you and the others that download and use the drivers and information here. I have coded quite a bit over the years but that is not my primary career field. In fact, when I started this project last year, I had to dust off my old C books. The complexity of these drivers is absolutely astounding. Laws from countries around the world have to be compiled with and the various modes that are supported add complexity. I monitor the Linux Wireless mailing list to help me pull these drivers along as the kernel is updated which can be a full time job given I'm trying to keep 5 drivers alive and working as best I can. I do get help from others. Pull requests from kind souls seem to come in when I get behind. Testing and using the variety of drivers and adapters has given me a perspective that few others have. I can see the difference in the quality of the drivers and adapters. I can tell you which of the Realtek drivers is best for AP mode (8812au). I can tell you which is best if all you want is the fastest managed mode (88x2bu). 

@DeityOfChaos 

You are correct in your above statement about Aircrack-ng's 8812au driver. It is based on old code. The driver here is more modern and is vastly different internally. It is the best Realtek out-of-kernel driver I have ever seen. It is not perfect but it is by far the best of the 5 Realtek drivers that I have posted here.

I see you are getting an Alfa ACM. If you are mostly used to using Realtek adapters, there are some things to watch for with ACM:

It is using an in-kernel driver. As an in-kernel driver, it is Linux Wireless standards compliant or very close to it. When using iw and ip, you are going to see output that is somewhat different in some areas than you are used to seeing. You will also see differences when using wavemon. The Realtek drivers are not standards compliant in many areas. In fact, they provide bogus information a lot. When you look for driver parameters... the ACM driver only has one whereas the Realtek drivers have dozens. So you ask how do I tell the ACM driver to go to USB 3 mode? The answer is that you don't. If detects and goes to USB 3 mode on its own. I have yet to see a problem in this regard. It just works. Another big advantage of the ACM is how little power it uses. I think you mentioned you use a RasPi4b. I have an ACM and a SSD plugged into the USB 3 ports and no problems with power. The ACM uses from 180 mA to 380 mA when pushed per my meter. That is really good. I hope you enjoy your ACM as much as I do mine and I would like to get a report from you if you don't mind. Could you put it in the issues at this repo?

https://github.com/morrownr/USB-WiFi

DeityOfChaos, 2021-03-16:
> @pavetrck
>
> I had a little time to look at your issue this evening. I have noticed this myself but have not been able to track down the cause. Something I have noticed is that the source we have for [...]

if with Bogus information you mean "Serial Number 123456" yea I have noticed it on Realtek stuff LOL

Thanks mate, my ACM and Cudy arrive tomorrow. In the meantime I just keep exploring 802.11 tools and code

Regarding "Laws from countries around the world have to be compiled with and the various modes that are supported add complexity" I didn't know that is part of the driver (i'll later look deeper into this repository to find the files on that matter). I intend to edit that off. I've edited my CRDA Database File to allow me to push to 30 dBm but I am yet to confirm if it works. Can't really trust values from software tools, I need to get something to measure the actual physical output power out of the SMA antenna interface

morrownr, 2021-03-17:
@DeityOfChaos 
@DeityOfChaos 

One of the files that addresses country related stuff is:

core/rtw_chplan.c

While I can appreciate your enthusiasm, let me temper it somewhat. Expect to break the driver if you mess around with that stuff. There is firmware that we can't access in this driver and on the adapters.  The best you can probably do is set an incorrect country code if you are looking to use channels or power settings that are not allowed in your country. And that is illegal. There are probably other ways to accomplish what you wish do... maybe directional antennas?

If you are hot on the idea of tracking down problems and finding fixes, let me suggest you look at the following issue:

https://github.com/morrownr/88x2bu/issues/47

WPA3 is simply not working in any of the drivers that I have up here. Realtek says WPA3 works. This is a challenge.

morrownr, 2021-03-25:
@pavetrck 

I've been busy working on some other issues in the limited time I have had available over the last week so am now trying to catch up on things. Did we solve the original issue or just make a mess? Should close this issue and let you refile if there is still a problem?

morrownr, 2021-03-27:
@DeityOfChaos 

Where is the report on that ACM you bought?

morrownr, 2021-04-01:
I going to close this issue as it looks like the original issue was solved. Any follow on issues can be posted in a new issue.

closed 2021-04-01 by morrownr