morrownr/8812au #17: Could not locate dkms.conf file - Ubuntu 18.04 / 5.4.80-hiveos

dan7298, 2021-03-13:
I'm stuck at this, tried a lot of things from over Google to no avail. Any help will be greatly appreciated!!!
```
Installing rtl8812au-5.9.3.2
Copying source files to: /usr/src/rtl8812au-5.9.3.2
Copying 8812au.conf to: /etc/modprobe.d

Creating symlink /var/lib/dkms/rtl8812au/5.9.3.2/source ->
                 /usr/src/rtl8812au-5.9.3.2

DKMS: add completed.

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.4.80-hiveos KSRC=/lib/modules/5.4.80-hiveos/build..............
cleaning build area...

DKMS: build completed.

8812au:
Running module version sanity check.

Good news! Module version v5.9.3.2_37279.20201012 for 8812au.ko exactly matches what is already found in kernel 5.4.80-hiveos.
DKMS will not replace this module.
You may override by specifying --force.

depmod...

DKMS: install completed.
dkms status:
Error! Could not locate dkms.conf file
File: does not exist.
The driver was installed successfully.
```

morrownr, 2021-03-15:
It is good to meet you. There is nothing wrong with being a newbie with Linux. I was a newbie with Linux back in 1994. It can be a lot of fun to run Linux as long as you don't lose your dkms.conf file. Okay, that was a joke and I'm sure the last thing you want is a joke right now.

Let's see about figuring out what is going on here. I need a clearer picture of what the situation is so here are some questions:

What is 5.4.80-hiveos? I don't recognize "hiveos."

Can you run the following and post the results?

$ dkms status

The installation script is telling us that you already have the driver installed. Which leads to a question about how the driver got installed? Can you clarify this?

Can you provide some information about you computer?

Can you tell me the brand and model of wifi adapter you have?

Please run the following and post the results:

$ lsusb

That should be enough to get started.

dan7298, 2021-03-15:
5.4.80-hiveos is a custom Linux kernel made by Hive for their [mining OS](https://hiveos.farm/)

**dkms status**
```
Could not locate dkms.conf file.
File:  does not exist.
```
**Installed the drivers through console as root:**
```
> sudo apt-get update
> sudo apt-get install -y dkms git
> mkdir src
> cd ~/src
> git clone https://github.com/morrownr/8812au.git
> cd ~/src/8812au
> sudo ./install-driver.sh
> sudo reboot
```
**System:**
```
Mobo:         B450 Pro4
CPU:          Athlon 3000G
RAM:          8GB DDR4 3200MHz
Storage:      256GB M.2 NVME
Wifi adapter: TP-Link T2U Nano V1.0
```
**adapter ID from lsusb:**
**2357:011e**

morrownr, 2021-03-15:
Okay, I see some things we need to talk about.

The Aircrack driver you referenced is not based on the 5.9.3.2 source code. It is based on the 5.6.4.2 source code and is relatively old. It is not clear to me if that repo is getting regular updates to keep the driver up to date with new kernel releases so you may want to steer clear of it.

The 5.9.3.2 source code for the rtl8812au used here is the latest release and is very modern and stable as far as Realtek out-of-kernel drivers are concerned. Maintenance is ongoing.

With that said, your device ID tells me that the chipset inside your wifi adapter is a rtl8811au. You are installing the wrong driver. The driver you should be installing is the 8821au. Go here:

https://github.com/morrownr

and click on 8821au.

I keep a file in each driver repo that is called `supported-device-IDs` and it shows the supported device IDs.

Now that you are pointed toward the correct driver, I cannot explain the missing dkms.conf issue. I could make some guesses but what you really need to do if installing the correct driver leads to the same problem is you need to bring the issue to the attention of the makers of the hiveos distro. I say this for a couple of reasons. First, I'm at the limit of distros that I can install and test with. I have a small lab with limited resources. I test with several distros right now but do not have the time or resources to add another right now. Without me being able to directly test on hiveos, I would only be guessing. Second, this site, including all 5 Realtek drivers and the informational repos average over 3000 hits per week. If the installation instructions and scripts were not very solid, we would see many problem reports related to installation. But we aren't seeing that. This leads me to think the makers of hiveos have done something that is causing the problem.

Let me know how it turns out and if there is anything I can do to help, I'll see you over in the 8821au repo.

One last thing: You mentioned that you are still fairly new to Linux. You might benefit from a trip over to this site:

 https://github.com/morrownr/USB-WiFi

Using adapters supported by in-kernel drivers eliminates all the muss and fuss with lost dkms.conf files and many other related issues.

dan7298, 2021-03-15:
Alright, thank you!

closed 2021-03-15 by dan7298