morrownr/8812au #19: Error! Could not locate dkms.conf file. File:  does not exist.

danithechief, 2021-03-22:
I have this error:

Error! Could not locate dkms.conf file.
File:  does not exist.

someone can help plz?

morrownr, 2021-03-22:
Run:

$ sudo ./remove-driver.sh

and then follow the installation instructions again.

The driver was already installed when you tried to install it. I'm not sure how that came about but I will look into reworking the installation script logic.

danithechief, 2021-03-22:
I know how. I have already installed some time before...

Thx bro i will try later

morrownr, 2021-03-27:
I have reworked and tested a new version of the installation script. It should prevent inadvertent installation attempts over previous installations.

danithechief, 2021-03-27:
You are the King💪
The reason why ive installed it twice is, because i tought it was an update.
Because on my zorin os the Driver dont work stable…

thx

closed 2021-11-24 by morrownr