morrownr/8812au #20: gcc: error: unrecognized command line option ‘-mlittle-endian’

OPSTime, 2021-03-27:
**os**
proxmox VE 6.3

**uname -a**
`Linux dragon 5.4.103-1-pve #1 SMP PVE 5.4.103-1 (Sun, 07 Mar 2021 15:55:09 +0100) x86_64 GNU/Linux`

**Error：
cat /var/lib/dkms/rtl8812au/5.9.3.2/build/make.**log
```
DKMS make.log for rtl8812au-5.9.3.2 for kernel 5.4.103-1-pve (x86_64)
Sat 27 Mar 2021 07:14:31 PM CST
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.4.103-1-pve/build M=/var/lib/dkms/rtl8812au/5.9.3.2/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.103-1-pve'
  CC [M]  /var/lib/dkms/rtl8812au/5.9.3.2/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8812au/5.9.3.2/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8812au/5.9.3.2/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8812au/5.9.3.2/build/core/rtw_io.o
gcc: error: unrecognized command line option ‘-mlittle-endian’; did you mean ‘-fconvert=little-endian’?
gcc: error: unrecognized command line option ‘-mlittle-endian’; did you mean ‘-fconvert=little-endian’?
make[2]: *** [scripts/Makefile.build:269: /var/lib/dkms/rtl8812au/5.9.3.2/build/core/rtw_cmd.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [scripts/Makefile.build:269: /var/lib/dkms/rtl8812au/5.9.3.2/build/core/rtw_security.o] Error 1
  CC [M]  /var/lib/dkms/rtl8812au/5.9.3.2/build/core/rtw_ioctl_query.o
gcc: error: unrecognized command line option ‘-mlittle-endian’; did you mean ‘-fconvert=little-endian’?
make[2]: *** [scripts/Makefile.build:271: /var/lib/dkms/rtl8812au/5.9.3.2/build/core/rtw_debug.o] Error 1
  CC [M]  /var/lib/dkms/rtl8812au/5.9.3.2/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8812au/5.9.3.2/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8812au/5.9.3.2/build/core/rtw_mlme.o
gcc: error: unrecognized command line option ‘-mlittle-endian’; did you mean ‘-fconvert=little-endian’?
make[2]: *** [scripts/Makefile.build:271: /var/lib/dkms/rtl8812au/5.9.3.2/build/core/rtw_io.o] Error 1
gcc: error: unrecognized command line option ‘-mlittle-endian’; did you mean ‘-fconvert=little-endian’?
gcc: error: unrecognized command line option ‘-mlittle-endian’; did you mean ‘-fconvert=little-endian’?
make[2]: *** [scripts/Makefile.build:271: /var/lib/dkms/rtl8812au/5.9.3.2/build/core/rtw_ioctl_query.o] Error 1
make[2]: *** [scripts/Makefile.build:271: /var/lib/dkms/rtl8812au/5.9.3.2/build/core/rtw_ioctl_set.o] Error 1
gcc: error: unrecognized command line option ‘-mlittle-endian’; did you mean ‘-fconvert=little-endian’?
make[2]: *** [scripts/Makefile.build:271: /var/lib/dkms/rtl8812au/5.9.3.2/build/core/rtw_ieee80211.o] Error 1
gcc: error: unrecognized command line option ‘-mlittle-endian’; did you mean ‘-fconvert=little-endian’?
make[2]: *** [scripts/Makefile.build:271: /var/lib/dkms/rtl8812au/5.9.3.2/build/core/rtw_mlme.o] Error 1
make[1]: *** [Makefile:1754: /var/lib/dkms/rtl8812au/5.9.3.2/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.103-1-pve'
make: *** [Makefile:2397: modules] Error 2
```

morrownr, 2021-04-01:
Did you run the following file during installation?

$ sudo ./raspi64.sh

morrownr, 2021-04-04:
What happened? It appears the OP tried to compile for ARM but is using a x86 based system. If you use an x86 based system, do not run the Raspberry Pi preparation scripts. I have changed the README to make this clear.

Fix? Run `$ sudo ./remove-driver.sh` , reboot and start over.

Regards.

OPSTime, 2021-04-09:
https://github.com/gnab/rtl8812au.git 
It is normal that the driver has been tested.

morrownr, 2021-04-09:
I don't understand your statement. Can you clarify?