morrownr/8812au #21: unstable driver with last kernel in kali x86

andreagonzalez2k, 2021-04-03:
Hello,
updated Kali x86 to latest version today:
Linux a 5.10.0-kali6-686-pae #1 SMP Debian 5.10.26-1kali2 (2021-04-01) i686 GNU/Linux
In previous versions the driver worked very well with no issues.
But in this version airodump-ng does not filter by channel and even it does not find any ap from time to time.
Using default options with a usb 3.0 device powered in a usb 2.0 port.
Attached dmesg output shows errors. They repeat in a loop.
[error_8812.txt](https://github.com/morrownr/8812au/files/6253169/error_8812.txt)

andreagonzalez2k, 2021-04-03:
Sorry, my fault. I forgot to kill conflicting apps with:
airmon-ng check kill
Now it works like a charm again.
Could it be possible not to have to kill Network manager?
Thanks for your driver, it is the best for my 8812 device,

morrownr, 2021-04-04:
You are welcome. 

Quote: "Could it be possible not to have to kill Network manager?"

Answer: Yes, I have been working on a new guide. It seems to work well here. Can I get you to test it and advise?

Test Monitor Mode

Setup:

Install the aircrack-ng package
```
$ sudo apt install aircrack-ng
```

Ensure Network Manager doesn't cause problems
```
$ sudo nano /etc/NetworkManager/NetworkManager.conf
```
add
```
[keyfile]
unmanaged-devices=interface-name:mon0;interface-name:mon1
```
Note: The above Network Manager to disregard two interfaces.

Enable monitor mode using iw and ip:
```
$ sudo iw dev
```
```
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr 00:c0:ca:ad:4b:cc
		type managed
		txpower 12.00 dBm
```
```
$ sudo ip link set wlan0 down
$ sudo iw phy phy0 interface add mon0 type monitor
$ sudo ip link set mon0 down
$ sudo iw mon0 set monitor control
$ sudo ip link set mon0 up
$ sudo iw dev mon0 set channel 36 (or whatever channel you want)
```
-----
Test injection
```
$ sudo aireplay-ng --test mon0
```
-----
Test deauth
```
$ sudo airodump-ng mon0 --band ag

$ sudo airodump-ng mon0 --bssid <MAC address of router> --channel <channel of router>

$ sudo aireplay-ng --deauth 0 -c <deviceMAC> -a <routerMAC> mon0 -D
```
-----

Please let me know how it goes and what changes I need to make.

andreagonzalez2k, 2021-04-04:
Hello, it seems that it does not work.

┌──(root💀a)-[/home/a/Documentos/temp]
└─# more /etc/NetworkManager/NetworkManager.conf    
[main]
plugins=ifupdown,keyfile

[ifupdown]
managed=false

[keyfile]
unmanaged-devices=interface-name:mon0;interface-name:mon1
                                                                                                                             
┌──(root💀a)-[/home/a/Documentos/temp]
└─# iw dev                                      
phy#0
        Interface mon0
                ifindex 4
                wdev 0x2
                addr 00:00:00:00:00:00
                type monitor
                channel 44 (5220 MHz), width: 20 MHz (no HT), center1: 5220 MHz
                txpower 15.00 dBm
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr 46:92:e2:0b:7b:2c
                type monitor
                channel 44 (5220 MHz), width: 20 MHz (no HT), center1: 5220 MHz
                txpower 15.00 dBm

┌──(root💀a)-[/home/a/Documentos/temp]
└─# aireplay-ng --test mon0                     
21:33:21  Trying broadcast probe requests...
21:33:23  No Answer...
21:33:23  Found 0 APs

Using "airodump-ng mon0 --band ag" it does not find any ap in any channel (a or g band).
After I executed "airmon-ng check kill" and aireplay-ng anf airodump-ng work again.
One more thing is that even when I kill NetworkManager and it works in dmesg there are two warnings. I attached the file with them.
[warnings.txt](https://github.com/morrownr/8812au/files/6255258/warnings.txt)

morrownr, 2021-04-04:
```
Interface wlan0
ifindex 3
wdev 0x1
addr 46:92:e2:0b:7b:2c
type monitor
```

How did wlan0 get to be type monitor? It should not be monitor.

Before executing the steps I posted, you really need a clean fresh boot and don't run airmon-ng.

andreagonzalez2k, 2021-04-04:
Hello, fresh boot. It seems that when executing "iw phy phy0 interface add mon0 type monitor" already puts mon0 in monitor mode. "iw mon0 set monitor control" also puts wlan0 in monitor mode.

┌──(root💀a)-[/home/a]
└─# more /etc/NetworkManager/NetworkManager.conf
[main]
plugins=ifupdown,keyfile

[ifupdown]
managed=false

[keyfile]
unmanaged-devices=interface-name:mon0;interface-name:mon1
                                                                                                                             
┌──(root💀a)-[/home/a]
└─# iw dev
                                                                                                                             
┌──(root💀a)-[/home/a]
└─# iw dev
phy#0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr ca:ad:93:b2:b1:db
                type managed
                txpower -100.00 dBm
                                                                                                                             
┌──(root💀a)-[/home/a]
└─# ip link set wlan0 down
                                                                                                                             
┌──(root💀a)-[/home/a]
└─# iw phy phy0 interface add mon0 type monitor
                                                                                                                             
┌──(root💀a)-[/home/a]
└─# iw dev                                     
phy#0
        Interface mon0
                ifindex 4
                wdev 0x2
                addr 00:00:00:00:00:00
                type monitor
                txpower -100.00 dBm
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr ca:ad:93:b2:b1:db
                type managed
                txpower -100.00 dBm
                                                                                                                             
┌──(root💀a)-[/home/a]
└─# ip link set mon0 down
                                                                                                                             
┌──(root💀a)-[/home/a]
└─# iw mon0 set monitor control
                                                                                                                             
┌──(root💀a)-[/home/a]
└─# iw dev                     
phy#0
        Interface mon0
                ifindex 4
                wdev 0x2
                addr 00:00:00:00:00:00
                type monitor
                channel 1 (2412 MHz), width: 20 MHz (no HT), center1: 2412 MHz
                txpower 17.00 dBm
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr ca:ad:93:b2:b1:db
                type monitor
                channel 1 (2412 MHz), width: 20 MHz (no HT), center1: 2412 MHz
                txpower 17.00 dBm
                                                                                                                             
┌──(root💀a)-[/home/a]
└─# ip link set mon0 up
                                                                                                                             
┌──(root💀a)-[/home/a]
└─# iw dev mon0 set channel 44
                                                                                                                             
┌──(root💀a)-[/home/a]
└─# aireplay-ng --test mon0
00:40:55  Trying broadcast probe requests...
00:40:57  No Answer...
00:40:57  Found 0 APs

morrownr, 2021-04-06:
Okay, I was testing with an adapter that has a mt7612u chipset on a Linux Mint box. Let me see about setting up an adapter with a rtl8812au chipset on a Kali box to see what the difference is with the Realtek driver.

Yes, I am aware that some of the lines are not needed in some situations. I try to avoid using "airmon-ng check kill" myself. The setup with mon0 works well here with my main box but let me check it in your setup. Will report back.

closed 2021-11-24 by morrownr