morrownr/8812au #24: Installation issue: An error occurred. dkms build error = 10

MaciejStawiar, 2021-05-08:
Hi All,
I have an issue with the installation. Doing well until running installation script:
`sudo ./install-driver.sh`
As a result I got:
```
Running: install-driver.sh version 20210421
Start installation.
Copying source files to: /usr/src/rtl8812au-5.9.3.2
Copying 8812au.conf to: /etc/modprobe.d

Creating symlink /var/lib/dkms/rtl8812au/5.9.3.2/source ->
               /usr/src/rtl8812au-5.9.3.2

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.10.17-v7l+ KSRC=/lib/modules/5.10.17-v7l+/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.17-v7l+ (armv7l)
Consult /var/lib/dkms/rtl8812au/5.9.3.2/build/make.log for more information.
An error occurred. dkms build error = 10
Please report this error.
```

I have RPi 4B with Raspbian GNU/Linux 10 (buster), went for 32bit with `sudo ./raspi32.sh`

NIC is Comfast CF-912AC: `Bus 001 Device 003: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter`

Logfile `/var/lib/dkms/rtl8812au/5.9.3.2/build/make.log` output:
```
pi@raspberrypi:~/src/8812au $ cat /var/lib/dkms/rtl8812au/5.9.3.2/build/make.log
DKMS make.log for rtl8812au-5.9.3.2 for kernel 5.10.17-v7l+ (armv7l)
sob, 8 maj 2021, 10:58:34 CEST
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.10.17-v7l+/build M=/var/lib/dkms/rtl8812au/5.9.3.2/build  modules
make[1]: Wej\u015bcie do katalogu '/home/pi/linux-3a33f11c48572b9dd0fecac164b3990fc9234da8'

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

make[1]: *** [Makefile:718: include/config/auto.conf] B\u0142\u0105d 1
make[1]: Opuszczenie katalogu '/home/pi/linux-3a33f11c48572b9dd0fecac164b3990fc9234da8'
make: *** [Makefile:2412: modules] B\u0142\u0105d 2

```

Thanks for help in advance

morrownr, 2021-05-08:
Hi @MaciejStawiar 

You said you "went for ./raspi32.sh" but you did not say if your version of RasPiOS is 32 bit or 64 bit. That is something we need to know. 

If your RasPiOS is 32 bit then we need to dig deeper for the problem. I haven't been seeing any problems with the 32 bit version of RasPiOS and this driver and this is a high traffic site.

On the other hand, if your RasPiOS is the 64 bit version, we have seen problems that we can't fix as the issues are header problems in RasPiOS so the position that I am taking is that we will look at the problems once the 64 bit version of RasPiOS is no longer in beta. So, for a solid system using this driver right now, please install the 32 bit version of RasPiOS.

Regards,

Nick

MaciejStawiar, 2021-05-08:
Hi,
uname shows I have armv7l installed - I found this is 32bit version. 
Freshly downloaded from an official RPi page: https://www.raspberrypi.org/software/operating-systems/ -> Raspberry Pi OS with desktop

morrownr, 2021-08-01:
Closing issue since it is solved.

closed 2021-08-01 by morrownr