morrownr/8812au #25: Access Point: Clinets do not get an IP address

k1moradi, 2021-05-09:
Hi there,

I used your instructions to install hostapd on ubuntu 18.04 installed on an old PC. I uninstalled network-manger, and installed dhcpcd5. Everything else was based on your instructions. The problem is the PC (server) itself has internet access, but the WiFi clients do not. They are able to connect but they do not get IP address.

morrownr, 2021-05-10:
Hi there yourself,

I am currently working on a Ubuntu/ amd64 version of the guide. I hope to have it available this coming week.

The problem is in the bridging. I'll try to get that worked out. The Raspberrry Pi OS and Ubuntu are enough different that it requires different guides.

Your usb adapter is based on the rtl8812au? If so, good. If not, then what is it? The reason I ask is that many of the Realtek drivers say they can do AP mode but they really don't do a good job. The 8812au is the exception. It really does do a good job of AP mode.

Nick

k1moradi, 2021-05-10:
That's very good news. 

My ubuntu is not really ubuntu anymore because I have modified it a lot, but I can test your guide.

Yes, I do have 8812au (1200 Mbps).  My card can do bridging with your version of the driver, but I had problem bridging with the previous versions (like Aircrack's version). Currently, I use network-manager to make my hotspot. It works, I could make a very stable wifi 5 hotspot, but the speed is limited to 176 Mbps (tested channel 48 only), the last time I checked. Also, in network-manager devices connected to the same hotspot are unable to see each other, which is problematic for Chromecast and Airplay functioning. With your hostapd guide, I could make a hotspot with 576 Mbps speed but clients do not get an IP. Even with the network-manager, whenever I bridge eth with wlan devices unable to get an IP. (I worked all this weekend to fix the problem but I could not).

I was wondering is there a way to increase the network-manager hotspot speed?

morrownr, 2021-05-10:
I am making good progress on the new Ubuntu guide today but it may be a couple of days before I am ready for it to be tested. It should provide a stable 80 Mhz channel width capability. It appears the key is properly disabling Network Manager and going fully with a properly setup systemd-networkd configuration. 

FWIW, the Aircrack version of the driver is based on an older version of the driver from Realtek. When I started work on this version, it became apparent to me quickly that this version is a dramatically better driver than the old version. I'm no fan of Realtek out-of-kernel drivers but this driver is by far the best driver I have ever seen from Realtek. The other Realtek drivers do not compare at all. Now, that is not to say that it is a great driver. The overall best driver for AC1200/AC1300 class USB WiFi on Linux is the mt7612u and it is not close.

k1moradi, 2021-05-10:
WoW, I wished I had one of mt7612u, but I'm sure 8812au will get better. I hope, it is mainlined in the kernel someday.

Good luck with the new guide! Please let me know whenever you needed an Ubuntu tester.

k1moradi, 2021-05-12:
1 and 2) Thanks for sharing this guide and the repo with me! 

I have ubuntu 18.04. I tested the guide. It is well written. I suggest masking the network manager as the network-manager-gnome applet was able to start network-manager in my case. 

Is networking service (/etc/network/interfaces) is still needed after enabling systemd-networkd?

Is it possible to combine all systemd-networkd config files in on file?

Unfortunately, I could not make a functioning hotspot. The problem was not with your guide. 

Let me explain. I have an old spare Dell Inspiron 6400 laptop (core 2 duo CPU with 4 GB of ram and USB 2 ports) that I intend to use as a router/hotspot/media server. I have two eth interfaces: one of them is a USB dongle connected to the Internet modem, and the other one is a built-in eth connected to the TV. I also have AC1200 wifi acting as an access point.

The computer is connected to the Internet with a Spectrum (the Internet company) modem. The modem is acting very peculiar. For instance, whenever I connect it to a different eth (or boot to a different OS with the same eth) interface I need to restart the modem to be able to connect to the Internet. Otherwise (i.e., without a restart), the connection is not going to be established. Whenever, I bridge two eth interfaces (or eth with wlan), even after restarting the modem I am unable to get a connection. Likewise, bridge members like wlan are unable to get an IP.

However, the network manager allows functional Internet sharing without making a bridge connection. I have no idea how because I have limited knowledge of networking.

morrownr, 2021-05-12:
Quote: " I suggest masking the network manager as the network-manager-gnome applet was able to start network-manager in my case."

Reply: Good idea. I did not think of that.

Quote: "Is networking service (/etc/network/interfaces) is still needed after enabling systemd-networkd?"

Reply: I am pretty sure it is not needed. I think, though I am not an expert yet, that systemd-network is basically a self contained networking management system. It is not full featured yet like Network Manager is but almost all distros use systemd whereas not all distros use Network Manager... and that is the reason I am using it. The RasPiOS does not use Network Manager.

Quote: "Is it possible to combine all systemd-networkd config files in one file?"

Reply: I will have to investigate further. I did make some modes today to the file names to clarify the order in which they are processed.

I have to admit that my head hurt after reading what you have and what you are trying to do. This might be easier to do with a RasPi4b and the RasPiOS. Have you considered that? You could then let the Dell Laptop be a laptop but it might be best to use a distro that does not use Gnome. Just some thoughts.

Nick

kmoradi-ucla, 2021-05-12:
Thanks for the suggestions. I might try OpenWrt first or buy a router that supports OpenWrt.

One thing, networking service (/etc/network/interfaces) is not part of the network manager. It is a different service (sudo systemctl stop networking).

morrownr, 2021-05-12:
I use two systems as Dumb APs (they could also be full up wifi routers). My fiber modem/router serves as the main router in my network. I use OpenWrt on a ZyXEL NBG6817 and hostapd on a RasPi4b with an Alfa AWUS036ACM usb wifi adapter. Both are very good. The RasPi setup is a lot cheaper and more flexible.

Thanks for saying something about the networking.service. It think it is a one time shot service and I need to fully investigate how it fits into the scenario. I had looked at it and other things many moons ago when originally looking at a setup for the RasPi but there are so many moving parts that are hard to keep up with.

k1moradi, 2021-05-13:
I'll look into them. How do you feel about one of these WiFi 6 devices. A few of them are based on the driver you like.
https://openwrt.org/toh/views/toh_extended_all?dataflt%5BWLAN+5.0GHz*%7E%5D=ax

morrownr, 2021-05-13:
The Linksys E8450 and the Belkin RT3200 are the same device with different stickers as far as I can tell. I've heard some good things about them but they are new so I guess we will see. They run the mt7915 chipset. That chipset is available in usb format so I'm hoping we see a usb wifi adapter based on it one of these days.

I follow the linux-wireless list to keep up with the happenings in order to try to maintain these Realtek out-of-kernel drivers. I like what I see from Intel, Atheros and Mediatek. The problem is that Intel and Atheros aren't releasing new chips for USB applications.

closed 2021-08-01 by morrownr