morrownr/8812au #27: ubuntu 21.04 - chipset 8811au - Kernel 5.11.x. It worked 3 days and stopped

neuberfran, 2021-05-24:
Hi,

I've been trying to get my wifi dongle (8811au) back on ubuntu 21.04 for 3 days.
 
**Can anyone help**

neuberfran@ubuntu2104:~$ **nmcli con** 
NAME               UUID                                  TYPE      DEVICE 
Conexão cabeada 1  4f0a34b8-d394-3914-b110-164598dfb94a  ethernet  enp3s0 
pablo              4a2dca15-84ed-4fba-837a-a732957991f1  wifi      --     
neuberfran@ubuntu2104:~$ **nmcli con up pablo**
Erro: Ativação da conexão falhou: No suitable device found for this connection (device lo not available because device is strictly unmanaged).
neuberfran@ubuntu2104:~$ **iwconfig**
lo        no wireless extensions.

enp3s0    no wireless extensions.

**enxd03745db188a**  **unassociated**  Nickname:"<WIFI@REALTEK>"
          Mode:Auto  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=0/100  Signal level=0 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

neuberfran@ubuntu2104:~$ **route -n**
Tabela de Roteamento IP do Kernel
Destino         Roteador        MáscaraGen.    Opções Métrica Ref   Uso Iface
0.0.0.0         192.168.1.1     0.0.0.0         UG    100    0        0 enp3s0
169.254.0.0     0.0.0.0         255.255.0.0     U     1000   0        0 enp3s0
192.168.1.0     0.0.0.0         255.255.255.0   U     100    0        0 enp3s0

morrownr, 2021-05-24:
Good morning @neuberfran 

Can you tell me where you got the driver you are using? It isn't the driver from this repo.

Regards,

Nick

morrownr, 2021-05-24:
Did not intend to chase you off. Knowing what driver you have installed would help me offer advice. If it is the 8812au driver in the Ubuntu repo, stay away from it. It is old and causes problems. If it is the one from aircrack-ng, it is also old but uninstallation would be different.

The driver you want is the one here as it is the newest and by far the best driver for the 8812au. I have no idea why Realtek finally kicked out a half way decent driver for this chipset but they did. 

Nick

neuberfran, 2021-05-24:
hi, @morrownr. Yes, I stay trying aircrack-ng now. Funny, that in ubuntu 20.10 this chipset is working ok.My chipset is 8811au and not 8812au. But the reps for 8812au are known to work on 8811au. Any help from you is welcome.

morrownr, 2021-05-24:
Oh, well, if you are sure it is a 8811au chipset then you want this driver:

https://github.com/morrownr/8821au

You really don't want the old aircrack-ng or Ubuntu drivers.

There is a file in the 8821au repo I just provided the link for that has a file called:

supported-device-IDs

Open it and read it so as to check your chipset against the supported chipsets.

Remember to uninstall all old drivers.

Good luck.

P.S. I have several repos here:

https://github.com/morrownr

One that might be of interest is this one:

https://github.com/morrownr/USB-WiFi

...as it has a lot of information.

neuberfran, 2021-05-24:
@morrownr Bus 002 Device 005: ID 2357:011e TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano]
Unfortunately, my id is not on the list. You (or me) have a job to do.
[image]

morrownr, 2021-05-25:
Yes, your device ID is in the file. I just checked, I think there is some confusion going on. Go to the following link:

https://github.com/morrownr?tab=repositories

Take note that I have several repos available. Each one gives a short rundown of what it is for. The one you want is 8821au.

neuberfran, 2021-05-25:
@morrownr Sadness still reigns. I followed all the steps, but I was not successful. I'm still excited to debug and keep trying.
[image]
[image]

morrownr, 2021-05-25:
The 8821au driver, this repo, installed successfully. Congrats. Now, it looks to me like another previously installed driver is also trying to do its thing with the same adapter. That is not good.

Run the follow and post the results:

$ dkms status

neuberfran, 2021-05-25:
@morrownr  neuberfran@ubuntu2104:~$ **dkms status**
rtl8821au, 5.8.2.3, 5.11.0-17-generic, x86_64: installed

morrownr, 2021-05-25:
Okay. We are making progress. The driver that dkms is showing is the driver from here. That is the one we want. It will work.

There is another driver, 88XXau, that is loading and causing problems. It wasn't installed with dkms so we need to know what driver that is and how to uninstall it. Since I am not a mind reader, the ball is back in your court. How did you install 88XXau?

neuberfran, 2021-05-25:
@morrownr **lsmod**

```
Module                  Size  Used by
nls_iso8859_1          16384  1
input_leds             16384  0
joydev                 28672  0
snd_hda_codec_via      20480  1
snd_hda_codec_generic    81920  1 snd_hda_codec_via
ledtrig_audio          16384  1 snd_hda_codec_generic
snd_hda_codec_hdmi     65536  1
snd_hda_intel          53248  3
snd_intel_dspcfg       28672  1 snd_hda_intel
soundwire_intel        40960  1 snd_intel_dspcfg
soundwire_generic_allocation    16384  1 soundwire_intel
soundwire_cadence      32768  1 soundwire_intel
snd_hda_codec         147456  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_via
snd_hda_core           94208  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_via
snd_hwdep              16384  1 snd_hda_codec
soundwire_bus          77824  3 soundwire_intel,soundwire_generic_allocation,soundwire_cadence
snd_soc_core          294912  1 soundwire_intel
hid_generic            16384  0
intel_rapl_msr         20480  0
snd_compress           28672  1 snd_soc_core
intel_rapl_common      28672  1 intel_rapl_msr
ac97_bus               16384  1 snd_soc_core
nouveau              2023424  15
usbhid                 61440  0
snd_pcm_dmaengine      16384  1 snd_soc_core
hid                   135168  2 usbhid,hid_generic
snd_pcm               118784  8 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,soundwire_intel,snd_compress,snd_soc_core,snd_hda_core,snd_pcm_dmaengine
x86_pkg_temp_thermal    20480  0
intel_powerclamp       20480  0
mxm_wmi                16384  1 nouveau
drm_ttm_helper         16384  1 nouveau
snd_seq_midi           20480  0
coretemp               20480  0
ttm                    73728  2 drm_ttm_helper,nouveau
snd_seq_midi_event     16384  1 snd_seq_midi
snd_rawmidi            36864  1 snd_seq_midi
kvm_intel             286720  0
drm_kms_helper        245760  1 nouveau
snd_seq                73728  2 snd_seq_midi,snd_seq_midi_event
cec                    53248  1 drm_kms_helper
rc_core                57344  1 cec
snd_seq_device         16384  3 snd_seq,snd_seq_midi,snd_rawmidi
kvm                   835584  1 kvm_intel
i2c_algo_bit           16384  1 nouveau
fb_sys_fops            16384  1 drm_kms_helper
syscopyarea            16384  1 drm_kms_helper
snd_timer              40960  2 snd_seq,snd_pcm
sysfillrect            16384  1 drm_kms_helper
sysimgblt              16384  1 drm_kms_helper
crct10dif_pclmul       16384  1
snd                    94208  19 snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_timer,snd_compress,snd_hda_codec_via,snd_soc_core,snd_pcm,snd_rawmidi
at24                   24576  0
ghash_clmulni_intel    16384  0
aesni_intel           372736  0
crypto_simd            16384  1 aesni_intel
cryptd                 24576  2 crypto_simd,ghash_clmulni_intel
soundcore              16384  1 snd
glue_helper            16384  1 aesni_intel
eeepc_wmi              16384  0
rapl                   20480  0
asus_wmi               40960  1 eeepc_wmi
sparse_keymap          16384  1 asus_wmi
8821au               1957888  0
intel_cstate           20480  0
wmi_bmof               16384  0
efi_pstore             16384  0
mac_hid                16384  0
cfg80211              892928  1 8821au
sch_fq_codel           20480  2
msr                    16384  0
parport_pc             45056  0
ppdev                  24576  0
lp                     20480  0
parport                65536  3 parport_pc,lp,ppdev
drm                   552960  10 drm_kms_helper,drm_ttm_helper,ttm,nouveau
ip_tables              32768  0
x_tables               49152  1 ip_tables
autofs4                45056  2
crc32_pclmul           16384  0
r8169                  94208  0
ahci                   40960  2
i2c_i801               32768  0
libahci                36864  1 ahci
lpc_ich                24576  0
i2c_smbus              20480  1 i2c_i801
realtek                28672  1
wmi                    32768  4 asus_wmi,wmi_bmof,mxm_wmi,nouveau
video                  49152  2 asus_wmi,nouveau
```

neuberfran, 2021-05-25:
@morrownr  Not solved. I came to think that the problem may be with my ubuntu 21.04 installation. But I am not available to reinstall now. In addition to what would mess with GRUB2 which is very correct and linked to ubuntu 20.10 and Open Core (hackintosh BS 11.1)****

morrownr, 2021-05-25:
That extra usb wifi driver seems to be gone. Still no wifi?

Have you tried moving the adapter to another usb port?

What does ( $ iw dev ) say?

You might be better off checking on the Ubuntu forums with installation issues.

neuberfran, 2021-05-25:
@morrownr iw dev
iw dev 
phy#0
	Interface wlxd03745db188a
		ifindex 3
		wdev 0x1
		addr d0:37:45:db:18:8a
		type managed
		txpower 12.00 dBm

morrownr, 2021-05-25:
You have a wifi interface. So... what's the problem?

neuberfran, 2021-05-25:
@morrownr I don't have pablo ssid. Look photo below from ubuntu 20.10.
[image]

morrownr, 2021-05-26:
Okay but have you selected it and put in the passphrase? It doesn't happen automatically until you have used it once.

neuberfran, 2021-05-26:
@morrownr  passphrase? It's not possible. Both via cli command and via gui command (dashboard)
[image]
[image]

morrownr, 2021-05-26:
So... what could cause this?

The driver is installed properly and you have a wifi interface. Let's try these:

$ sudo iw reg get

$ sudo rfkill unblock wlan

neuberfran, 2021-05-26:
@morrownr

morrownr, 2021-05-27:
Go to the main page of this repo and scroll down until your to the following section:

Set regulatory domain to correct setting in OS

Read it, do it and reboot.

neuberfran, 2021-05-27:
@morrownr not solved yet with commands sudo iw reg set BR  and
sudo nano /etc/default/crda  (REGDOMAIN=BR) and restart PC.

Do you have good knowledge of NetworkManager?

morrownr, 2021-05-27:
My knowledge of Network Manager is not that high. I spend a lot of my time in distros that don't use it... like RasPiOS.

I like to solve problems but this one is probably to the point where posting in...

https://ubuntuforums.org/

...in the Networking and Wireless forum is going to get better results.

What I know is that the driver is loading and a wifi network interface is being created. Why network manager is not finding your ap is unknown.

neuberfran, 2021-05-28:
@morrownr  https://askubuntu.com/questions/1341262/dongle-wifi-tp-link-ac600-archer-t2u-8811au-not-working-in-ubuntu-21-04

morrownr, 2021-05-28:
Quote:

lsmod

8821au 1957888 0 cfg80211 892928 2 8821au,88XXau

Reply: That 88XXau is still installed and is probably causing problems because that driver includes support for 8811au. The Aircrack driver has drivers for multiple chipsets in one driver. The drivers here don't do that.

I thought we got that 88XXau uninstalled. I am confused.

neuberfran, 2021-05-29:
@morrownr 
neuberfran@ubuntu2104:~$ **sudo rmmod 88XXau**
[sudo] senha para neuberfran: 
neuberfran@ubuntu2104:~$ **lsmod 88XXau**
Usage: lsmod
neuberfran@ubuntu2104:~$ **dkms status**
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/rtl8821au/5.8.2.3/source/dkms.conf does not exist.
neuberfran@ubuntu2104:~$ ls **/usr/src**
linux-headers-5.11.0-16  linux-headers-5.11.0-16-generic  linux-headers-5.11.0-17  linux-headers-5.11.0-17-generic
neuberfran@ubuntu2104:~$

neuberfran, 2021-05-29:
@morrownr Whenever I restart the PC the 88XXau appears again

morrownr, 2021-05-29:
88XXau appears again because it is not being uninstalled.

Let's see the results of the following again:

$ sudo dkms status

FWIW: This thread is a great example of the problems that Linux users have with Realtek chipset based USB WiFi adapters. Good alternatives are available:

https://github.com/morrownr/USB-WiFi

neuberfran, 2021-05-29:
@morrownr sudo dkms status
[sudo] senha para neuberfran: 
sudo: dkms: not found command

I deleted things manually. Including dkms. I want to start 8821au again. But first, I need to clean up 88XXau and the 8821au itself from the lsmod list

neuberfran, 2021-06-02:
@morrownr sudo dmesg |grep core
I have good news and bad news. The good thing is that (as per the photo of the **sudo dmesg |grep core** command) I only have the 8821au module installed now. I managed to uninstall them all. The bad thing is that I still have no wifi and no wlan0.
[image]

morrownr, 2021-06-02:
@neuberfran 

I do enjoy solving a good mystery but I think it is time to step back and assess the situation. At this point, if I were in your shoes, I would save my important documents to an external drive and do a fresh clean installation of Ubuntu 21.04. There are a couple of ways to get the Ubuntu installer to format the installation partition during installation.

I think earlier in this thread I mentioned that sudo is a weapon of mass destruction. Well, it is. I think the current situation should just be written off as a learning experience and a fresh start is what is called for.

Something else to discuss while looking at the situation. I work on these Realtek driver as a way to help my fellow Linux users. That does not mean that I like how Realtek does it. I think the way Realtek does their usb wifi support for Linux is terrible. I also have and use usb wifi adapters that use Mediatek and Atheros chipsets. The drivers for these adapters are maintained in the kernel and are Linux Wireless standards compliant. Using Mediatek chipset based adapters on Linux is like a breathe of fresh air. I also maintain a repo with information about adapters that work with in-kernel drivers.

https://github.com/morrownr/USB-WiFi

I invite you to take a look. There are currently 69 links to adapters that "just work."

neuberfran, 2021-06-02:
You need to learn more about networkmanager

morrownr, 2021-06-03:
I agree but there are a lot of things I need to learn more about and I am working on them. Has knowledge about Network Manager fixed this problem?

neuberfran, 2021-06-03:
@morrownr no yet bro, sorry

morrownr, 2021-08-01:
Status?

closed 2021-08-01 by morrownr