morrownr/8812au #31: Linksys WUSB6300 V2 uses rtl88x2bu

swhite39, 2021-07-16:
Linksys recently created a V2 of their WUSB6300 Dongle (Device ID 0x0045) that is based on the RealTek 8822BU chipset and therefore uses the 8822bu driver. The original v1 WiFi dongle (Device ID 0x003F) uses the 8812au chipset. 

I was able to get the V2 Dongle working with the 5.4 kernel and a custom build of your 88x2bu driver repository. 

If you have the time it is probably worth denoting the V1/V2 difference in the supported devices list for these drivers. I know you can get it via modinfo, but it's nice having information spoon fed some times ;)

Anyhow, thanks for posting all the realtek drivers and for your writeup about WiFi Dongles that work with Linux. You helped me out a lot!

morrownr, 2021-07-28:
Thanks for the input. I have changed both README's to reflect difference.

Quote: "I was able to get the V2 Dongle working with the 5.4 kernel and a custom build of your 88x2bu driver repository."

Did you have to add the device id to the 88x2bu driver before it would work? If so, please forward what you added so that I can add it to the 88x2bu repo.

morrownr, 2021-07-28:
I checked the 88x2bu repo and "0x13b1, 0x0045" was already in place so I'm curious as to why the custom build?

I added V2 to the README.

swhite39, 2021-07-28:
Sorry for not being clear, by custom build I was referring to some changes I made to the Makefile to build for my architecture. I am using a unique cross compilation toolchain targeting a Xilinx Zynq UltraScale Plus. I didn't have to edit any source code.

I did however try adding the device ID for the V2 to the 8812au driver. I go the driver to claim the device but things obviously didn't work. It took me a while to figure out the V2 redesign was using a different Realtek chipset. I wish that information was more readily available.

I found this site helpful for identifying the underlying chipset based on the vendor and product ID, but I guess the V2 is so new there was no information about it.
https://devicehunt.com

morrownr, 2021-07-28:
I changed the README for this driver and the 88x2bu driver. I keep a file called "supported-device-IDs" in each repo to help.

Here is a little free advice from one Linux user to another: NEVER order or buy a usb wifi adapter without knowing what chipset is inside first. Never. The quality of these Realtek out-of-kernel drivers varies greatly from one chipset to another. Currently the 8812au driver (here) is by far the best driver and the driver for the 8814au is the worst. The others fall somewhere in between.

I currently recommend Linux users give priority to buying adapters with mt7612u chipsets. I also currently recommend Linux stay away from adapters with rtl8814au and rtl8811au chipsets as the drivers are poor quality and there is no indication they will ever be updated...

closed 2021-11-24 by morrownr