morrownr/8812au #32: Alfa AWUS036ACM (mt7612u/mt7612un) vs. Alfa AWUS036ACHM (mt7610u)

personaingrata, 2021-07-25:
first of all, this is no issue, the driver works well.
i own the AWUS036AC, AWUS036ACH and AWUS036NHA. while the rtl8812au based adapters work very well in monitor mode hearing other access points, i'm disappointed with the signal strength in AP mode. the ar9271 based adapter delivers a more stable and slightly stronger signal while drawing much less power (measurings with the provided antennas). tx power settings, regdb hacks are ignored by the rtl8812au adapter and didn't bring me any good outcome. i tried different drivers over the years and your driver works best, thank you for that! i'm looking forward to replace these 2 adapters and would like to try one of your recommended mediatek chipset based adapters.

how does the AWUS036ACM and AWUS036ACHM compare to each other? i'm assuming both work well while in monitor mode, but how about signal strength in AP mode? according to your tests the AWUS036ACHM draws a little bit more mA under load, which leads me to the questions does the AWUS036ACHM deliver a stronger signal than the AWUS036ACM? have you compared their signal strength and has tx power setting an effect? WPA3 support is already a big plus, even though it's currently not working properly on my linux notebook, but this should only be a matter of time.

would they deliver a stronger signal in AP mode than the AWUS036AC/H?

i'm looking forward to hear from you, thanks for your time.

morrownr, 2021-07-25:
Quote: "how does the AWUS036ACM and AWUS036ACHM compare to each other?"

Reply: For my uses, I prefer the ACM over the ACHM. The ACM is an AC1200 class adapter whereas the ACHM is an AC600 class adapter. The ACM is wonderful in 5 GHz AP mode. In fact it is the best all around usb wifi adapter I have ever used.

I can do some testing here if you tell me what you are looking for. I have numerous adapters and it doesn't take long to switch out.

My experience with adapters that use the rtl8812bu chipset is that they work well in managed mode but not that great in monitor or master modes.

Regarding WPA3: Some distros do support WPA3 now (Ubuntu 21.04 as an example). I think by mid-2022 we will see all of the major distros' current LTS versions supporting WPA3.

personaingrata, 2021-07-26:
thanks for the quick reply,
i'm primarily interested in signal strength in ap mode 2G/5G, monitor mode performance/sensitivity and injection. download speeds aren't important to me.
so maybe you could just run hostapd and compare the signal strength with the rtl8812au based AC or AC and a quick run in airodump-ng and aireplay-ng for injection?

the nice thing about the ACHM is that it probably comes with a clip for attaching it to the display i haven't found something similar online for the small AC/ACM.

i haven't investigated the WPA3 issue yet, NetworkManager (1.32.4-1) (wpa_supplicant 2.9-8) reports that i'm connected via WPA3 but my router reports that the machine is connected via WPA2, but i'm not in a hurry since there's PMF.

morrownr, 2021-07-27:
The ACHM does come with a clip but 3M Command Strips work well with the ACM.

I have the following AC1200 Alfa adapters: ACHM (mt7610u), ACH (rtl8812au) and ACM (mt7612u).

[image]

I do a lot of comparison testing so here are some general observations regarding the ACHM and ACM:

The most sensitive (range) is the ACHM followed by the ACM. Both are well above average in this regard.

Regarding monitor mode performance, I rank them in this order: ACM, ACHM. Both are very good. The Mediatek drivers are Linux Standards compliant so they do monitor mode by the book.

Regarding injection: I am not a professional pen tester but I do injection testing. I suspect you would be happy with either the ACM or ACHM.

Regarding AP Mode: I haven't tested the ACHM in AP Mode as it is a AC600 class adapter and I usually only use AC1200 class adapters in AP Mode. What I can do is test the ACHM, ACM and ACH with hostapd over the next few days to gather data and post it. The only adapter we have in common is the ACH so if it is included, that should help you determine the relative capabilities.

Cheers

morrownr, 2021-08-01:
I haven't been able to test a wide range of situations in AP mode but I was able to set up and test the ACHM in one situation. I am concerned about a problem I am seeing but have not had time to fully explore it.

As it is right now, my recommendation would be the ACM. I have used the ACM extensively in all of the use cases you mentioned and I can say without reservation that it should meet your expectations. I can't say that about the ACHM at this point.

My recommendation is that you go with the ACM.

morrownr, 2021-08-01:
Oh, I did not mention signal strength in AP mode. I did compare the ACM to the ACH in AP mode. The readings I was seeing were about the same. Signal strength can be a difficult comparison because signal strength readings don't necessarily correlate with range and throughput. And then you throw in congestion, if you have any, and things get messy.  I've seen situations where reducing tx power helps range and throughput so you just have to figure out what works best in your situation.

If you go with the ACM, you will find advantages over the AC and ACH is some areas: It will do monitor mode by the book so utilities like aircrack-ng work like the instructions say they should work. You don't get that with any Realtek drivers. The ACM also supports WPA3 already and it supports things like interface combinations and extended features which aren't available with the Realtek drivers.

personaingrata, 2021-08-02:
thanks for your time and testing. i'm gonna order the ACM and experiment with it.
1 last question, is there anything i should consider regarding USB2/3 ports?

morrownr, 2021-08-02:
Regarding USB2/3: Nothing that comes to mind. The ACM will probably come with a USB3 extension cable. I've used the ACM in USB2 and USB3 ports. It is one of those things where there is not much to say. It "just works."

I did think of one thing to add. I have compared the ACM in AP mode 5 GHz to the 5 GHz on my Zyxel Armor Z2 AC2600  Wireless Router and I found that the range and signal quality are identical. The Zyxel, with Openwrt, will support DFS channels but the ACM will not in AP mode. I am not aware of any usb adapters that support DFS.

Let me know how it goes.

morrownr, 2021-08-23:
@personaingrata 

Quote: "how does the AWUS036ACM and AWUS036ACHM compare to each other?"

I have been testing the ACHM in AP mode for the last 2 weeks. I wish I had been able to give you an answer earlier but I had not used the ACHM in AP mode and I hate to do quick tests only to find out later that the information was not so good. Here goes:

I tested 2.4 GHz for one week and 5 GHz for one week. I compared the ACHM (MAX Power) to the ACH (MAX Power) and the ACM (High Power). The results were similar to previous testing in managed and monitor mode in that the ACHM showed better range than either the ACH or ACM. 

Average signal level tests showed the ACHM beating the ACH by 1 to 2 dBm.

Average signal level tests showed the ACHM beating the ACM by 6 to 10 dBm.

Actually usage confirmed the signal test results were good indications of longer range. The ACHM also showed stronger signal tests and range than my ZyXEL NBG6817 Wave 2 WiFi router. I also tested WPA3 capability. The ACHM is working well using WPA3 in all modes tested: managed, master and monitor. The ACM also tested good using WPA3 in managed, master and monitor modes. The ACH fell flat on its face when trying to use WPA3. When in managed mode, It won't even connect to an AP that is set to WPA2/WPA3 mixed mode. There is likely a total lack of support for 802.11w in the Realtek drivers that is contributing to this. I did note a lack of support for DFS channels in AP mode with ACHM but that is not surprising as I have yet to find a USB WiFi adapter that supports DFS channels in AP mode. I did grep around in the MT76 driver code in the kernel and, interestingly enough, the support appears to be there. I guess that is a future project.

Overall assessment: If your primary concern is range, the Alfa ACHM is the longest range dual band usb adapter I have ever tested. One thing to keep in mind is that it is an AC600 class adapter. It is solid and has long range but is not speedy. If you have 100 Mb/s internet service, it will certainly handle that without being the bottleneck but if you have internet service exceeding 200 Mb/s, it can be the bottleneck.

Nick

personaingrata, 2021-08-28:
i checked the ACH conncted to my FRITZ!Box and NetworkManager reports me WPA3 but when i check the connection on my router it says connected through WPA2. this applies to all my wireless adapters including my internal intel 7265, AC, ACH, ACM,.. but the connections work just fine.
the only time i get an actual WPA3 connection on my router is when i connect my android 11 phone.

if you want i can check how it behaves with hostapd on my machine, just send me your .conf file

morrownr, 2021-08-28:
Here is my 5GHz hostapd.conf: (Note: I do a lot of testing ... you have been warned)

# /etc/hostapd/hostapd-5g.conf
# Documentation: https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf
# 2021-08-28

# Defaults:
# SSID: myPI-5g
ssid=myPI-5g
# PASSPHRASE: myPW1234
wpa_passphrase=myPW1234
# Band: a - a = 5g (a/n/ac), g = 2g (b/g/n)
hw_mode=a
# Channel: 36
channel=36
# Country: US
country_code=US

# needs to match wireless interface in your system

# ACHM
#interface=wlxXX

# ACM
interface=wlxXX

# TEROW
#interface=wlxXX

# needs to match bridge interface name in your system
bridge=br0

driver=nl80211
#ctrl_interface=/var/run/hostapd
#ctrl_interface_group=0

# limit the frequencies used to those allowed in the country
ieee80211d=1

# enable DFS channels
#ieee80211h=1

#beacon_int=100
#dtim_period=2
#rts_threshold=2304
#fragm_threshold=2304

#max_num_sta=128
macaddr_acl=0

# security
auth_algs=1
#ignore_broadcast_ssid=0
# wpa=2 is required for WPA2 and WPA3 (read the docs)
wpa=2
rsn_pairwise=CCMP
# only one wpa_key_mgmt= line should be active.
# wpa_key_mgmt=WPA-PSK is required for WPA2-AES
#wpa_key_mgmt=WPA-PSK
# wpa_key_mgmt=SAE WPA-PSK is required for WPA3-AES Transitional
#wpa_key_mgmt=SAE WPA-PSK
# wpa_key_mgmt=SAE is required for WPA3-SAE
wpa_key_mgmt=SAE
#wpa_group_rekey=1800
# ieee80211w=1 is required for WPA-3 SAE Transitional
# ieee80211w=2 is required for WPA-3 SAE
ieee80211w=2
# if parameter is not set, 19 is the default value.
#sae_groups=19 20 21 25 26
# sae_require_mfp=1 is required for WPA-3 SAE Transitional
#sae_require_mfp=1
# if parameter is not 9 set, 5 is the default value.
#sae_anti_clogging_threshold=10

# Note: Capabilities can vary even between adapters with the same chipset.
#
# Note: Only one ht_capab= line and one vht_capab= should be active. The
# content of these lines is determined by the capabilities of your adapter.
#
# IEEE 802.11n
ieee80211n=1
#require_ht=1
wmm_enabled=1
#
# mt7610u
#ht_capab=[HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][RX-STBC-1]
# mt7612u
ht_capab=[LDPC][HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC-1]
#ht_capab=[GF][SHORT-GI-20]
#
# rtl8812au - rtl8811au -  rtl8812bu - rtl8811cu - rtl8814au
#ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
#

# IEEE 802.11ac
ieee80211ac=1
#require_vht=1
#
# mt7610u
#vht_capab=[SHORT-GI-80][RX-STBC-1][MAX-A-MPDU-LEN-EXP3][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]
# mt7612u
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][MAX-A-MPDU-LEN-EXP3][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]
#
# rtl8812au - rtl8811au -  rtl8812bu - rtl8811cu - rtl8814au
#vht_capab=[MAX-MPDU-11454][SHORT-GI-80][SU-BEAMFORMER][HTC-VHT]
# rtl8814au
#vht_capab=[MAX-MPDU-11454][RXLDPC][SHORT-GI-80][SU-BEAMFORMER][MAX-A-MPDU-LEN-EXP3][HTC-VHT]
# Note: [TX-STBC-2BY1] causes problems
#
# Required for 80 MHz width channel operation on band 2 - 5g
vht_oper_chwidth=1
#
# Use the next line with channel 36  (36 + 6 = 42) band 2 - 5g
vht_oper_centr_freq_seg0_idx=42
# Use the next line with channel 149 (149 + 6 = 155) band 2 - 5g
#vht_oper_centr_freq_seg0_idx=155
# Use the next line with channel 132 (132 + 6 = 138) band 2 - 5g
#vht_oper_centr_freq_seg0_idx=138

# Event logger - as desired
#logger_syslog=-1
#logger_syslog_level=2
#logger_stdout=-1
#logger_stdout_level=2

# WMM - as desired
#uapsd_advertisement_enabled=1
#wmm_ac_bk_cwmin=4
#wmm_ac_bk_cwmax=10
#wmm_ac_bk_aifs=7
#wmm_ac_bk_txop_limit=0
#wmm_ac_bk_acm=0
#wmm_ac_be_aifs=3
#wmm_ac_be_cwmin=4
#wmm_ac_be_cwmax=10
#wmm_ac_be_txop_limit=0
#wmm_ac_be_acm=0
#wmm_ac_vi_aifs=2
#wmm_ac_vi_cwmin=3
#wmm_ac_vi_cwmax=4
#wmm_ac_vi_txop_limit=94
#wmm_ac_vi_acm=0
#wmm_ac_vo_aifs=2
#wmm_ac_vo_cwmin=2
#wmm_ac_vo_cwmax=3
#wmm_ac_vo_txop_limit=47
#wmm_ac_vo_acm=0

# TX queue parameters - as desired
#tx_queue_data3_aifs=7
#tx_queue_data3_cwmin=15
#tx_queue_data3_cwmax=1023
#tx_queue_data3_burst=0
#tx_queue_data2_aifs=3
#tx_queue_data2_cwmin=15
#tx_queue_data2_cwmax=63
#tx_queue_data2_burst=0
#tx_queue_data1_aifs=1
#tx_queue_data1_cwmin=7
#tx_queue_data1_cwmax=15
#tx_queue_data1_burst=3.0
#tx_queue_data0_aifs=1
#tx_queue_data0_cwmin=3
#tx_queue_data0_cwmax=7
#tx_queue_data0_burst=1.5

# end of hostapd-5g.conf

morrownr, 2021-08-31:
Here is a command line option to help see what mode you are using:

$ sudo wpa_cli status

personaingrata, 2021-09-04:
had some time today and i had no luck connecting via your hostapd.conf, no matter which adapter i tried.
finally my problem seems to be fixed, i can now connect my laptop via WPA3 to my router

morrownr, 2021-09-05:
> had some time today and i had no luck connecting via your hostapd.conf, no matter which adapter i tried.
So many settings depend on the platform, specific distro and specific adapter. I maintain a guide called Bridged Wireless Access Point at the following site:

https://github.com/morrownr/7612u

> finally my problem seems to be fixed, i can now connect my laptop via WPA3 to my router

Great.

morrownr, 2021-09-07:
Another way to check security:

$ nmcli device wifi list

        XX:59:73:FE:8B:F4  Trout                               Infra  11   260 Mbit/s  99      ▂▄▆█  WPA2             
        XX:C0:CA:AD:CB:84  myPI-2g                        Infra  6     130 Mbit/s  97      ▂▄▆█  WPA3