morrownr/8812au #33: Alfa AWUS036AC as Access point and dual band causes hostapd not to start on Ubuntu 18.04 x86_64.

CpServiceSpb, 2021-07-25:
First of all, thanks a lot for the author of morrownr.
I built the driver, and tried to make 2 hostapd configurations - each for its band (2.4/5 GHz) by [Bridged_Wireless_Access_Point.md](https://github.com/morrownr/8812au/blob/5.9.3.2/Bridged_Wireless_Access_Point.md) on Ubuntu 18.04, not for Rasp, but for x86_64.
Driver stared well.
But I was unsuccessfull at starting of hostapd v 2.6.
There is the same interface used as for 2.4 GHz as for 5 GHz hostapd configuration.
And hostapd can not initiate the interface for the 2nd band configuration.
As I understood multiple SSID configuration is necessary.
For that, two different interfaces - by names and of course by MAC - have to be created.
For example - wlan2g with MAC 02:blah-blah:02 and wlan5g with MAC 02:blah-blah:05.
The first 02 is important changed value allowing administerimg MACs by hostapd.
So, may be te point is to add to the above instruction or may be I do somehing wrong.
May somebody clarify the situation ?

morrownr, 2021-07-25:
Welcome. The project you are working on can take some time to become familiar with all of the working parts. Try not to get discouraged.

The example I posted is based on the RaspPiOS for good reason. It is much easier to do this type of project with RaspPiOS. Ubuntu manages its networking differently, including the use of Network Manager. For this to work on Ubuntu, you either need to know a lot about Network Manager or just shut it down altogether. Did you shut Network Manager down?

It isn't clear to me that you understand that 2 wifi adapters are required for dual band. An adapter can be in AP mode for 2.4 GHz or 5 GHz but not both. Do you have 2 wifi adapters?

morrownr, 2021-07-26:
Oh. Something I just thought about...

You say you have a Alfa AWUS036AC. I'm pretty sure that adapter uses a 8812bu chipset. This repo is for the 8812au chipset so this driver will not work for the AWUS036AC. How, I have Linux drivers up for almost all of the modern Realtek based adapters. The one you want is https://github.com/morrownr/88x2bu

FYI: There is actually a x86 version of the Raspberry Pi OS... no kidding:

https://www.raspberrypi.org/software/raspberry-pi-desktop/

I'm not telling you what to do or even suggesting it, I am simply adding to your knowledge. It could be a good way to go. What hardware do you have and what provides the internet access to the hardware?

FWIW: If that Alfa adapter is based on the 8812bu chipset, that is not a good chipset to use for AP mode in my experience. I have worked on that driver until I am blue in the face and all I can say is Realtek did a good job with managed mode and that is it. IMHO, the best adapter for AP mode is the Alfa AWUS036ACM. More info available at these sites:

https://github.com/morrownr/USB-WiFi

https://github.com/morrownr/7612u

Building your own wifi router/access point/hotspot is really cool and can teach you a lot.

CpServiceSpb, 2021-07-26:
I have only an only one Alfa AWUS036AC USB dongle which is based exactly on RTL8812AU, you may see it here https://www.alfa.com.tw/products/awus036ac?variant=36473965838408.
No one others wifi I have.
Here is part of lsusb:
`0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter`
And the driver for 8812AU works.
So, get me to thinking that the adapter can not work on dual band simulteniously.
But I thought that it can be acst on both frquencies as some wi-fi routers which can be found by wifi clients (PCs, laptops, smartphones as at 2.4 GHz as 5 GHz) .
Moreover I read somewhere that some wifi adapter is able to operate at both frequencies simulteniously especially the devices with 2 or more antennas.
My wifi has 2 antennas.
Regarding OS, I have Ubuntu 18.04 x64 LTS server with no GUI with Network manager uninstalled as well as Netplan and with Ifupdown installed and operating.
And one more "situation" is with my AWUS036AC - it is plugged to USB3 port.
If I launch insmod 8812au.ko with rtw_switch_usb_mode=1, the adapter is unplugged itself.

morrownr, 2021-07-26:
Okay. I was wrong about the Alfa AWUS036AC. It does have the 8812au chipset. That is a good thing.

Quote: "But I thought that it can be acst on both frquencies as some wi-fi routers..."

Dual band WiFi routers have two radios inside. These WiFi adapters do not. If you have a link that shows wifi adapters that can work as dual band, please post the link. 

I'm going to recommend that we pick our battles carefully. How about you do a single band installation. You pick the band. We get that working and then so where we are. Starting with dual band increases the complexity greatly.

Okay. So the server version and netplan is shut down. Good.

Quote: "And one more "situation" is with my AWUS036AC - it is plugged to USB3 port."

Not all adapters work with all cables and cradles. Maybe we should go for function and speed it up later? For now, maybe 2 is the better setting.

Try to rework your installation to be a single band setup and then show me what you are seeing.

CpServiceSpb, 2021-07-27:
Unfortunatelly I got the response from Alfa support - no, the device do not support 2.4 GHz and 5 GHz at the same time.
So, I achieved success at 2.4 GHz - clients see AP, but during auth:
wlan: STA client MAC IEEE 802.11: associated
wlan: STA client MAC IEEE 802.11: disassociated
wlan: STA client MAC IEEE 802.11: associated
wlan: STA client MAC IEEE 802.11: disassociated
wlan: STA client MAC IEEE 802.11: associated
where client MAC is MAC client.
I will discover further.

morrownr, 2021-07-27:
Post the contents of your hostpad.conf

If doing 2.4g, recommend these module parameter settings:

rtw_vht_enable=1 rtw_switch_usb_mode=2

CpServiceSpb, 2021-07-29:
Here is:
```
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
interface=wlan
driver=nl80211
bridge=br0

### IEEE 802.11
ssid=MySuperPuper2G
# Band: a = 5g (a/n/ac), g = 2g (b/g/n)
hw_mode=g
channel=6
max_num_sta=128
auth_algs=1

### DFS
country_code=US
ieee80211d=1
ieee80211h=1

### IEEE 802.11n
ieee80211n=1
ht_capab=[HT20][HT40+][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40]

### IEEE 802.11ac
ieee80211ac=1
#vht_oper_chwidth=1
#vht_capab=[HT40-]
#vht_oper_centr_freq_seg0_idx=46

### IEEE 802.11i
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
wpa_passphrase=12345678

#ignore_broadcast_ssid=0
logger_syslog_level=0

### WMM
wmm_enabled=1
uapsd_advertisement_enabled=1
wmm_ac_bk_cwmin=4
wmm_ac_bk_cwmax=10
wmm_ac_bk_aifs=7
wmm_ac_bk_txop_limit=0
wmm_ac_bk_acm=0
wmm_ac_be_aifs=3
wmm_ac_be_cwmin=4
wmm_ac_be_cwmax=10
wmm_ac_be_txop_limit=0
wmm_ac_be_acm=0
wmm_ac_vi_aifs=2
wmm_ac_vi_cwmin=3
wmm_ac_vi_cwmax=4
wmm_ac_vi_txop_limit=94
wmm_ac_vi_acm=0
wmm_ac_vo_aifs=2
wmm_ac_vo_cwmin=2
wmm_ac_vo_cwmax=3
wmm_ac_vo_txop_limit=47
wmm_ac_vo_acm=0

### TX queue parameters
tx_queue_data3_aifs=7
tx_queue_data3_cwmin=15
tx_queue_data3_cwmax=1023
tx_queue_data3_burst=0
tx_queue_data2_aifs=3
tx_queue_data2_cwmin=15
tx_queue_data2_cwmax=63
tx_queue_data2_burst=0
tx_queue_data1_aifs=1
tx_queue_data1_cwmin=7
tx_queue_data1_cwmax=15
tx_queue_data1_burst=3.0
tx_queue_data0_aifs=1
tx_queue_data0_cwmin=3
tx_queue_data0_cwmax=7
tx_queue_data0_burst=1.5
```
At the time There is no br0 interface on the system.
And dhcp server which I have, does not listen on wlan interface.
But s I understand there is faultin on a stage before net level one.

morrownr, 2021-07-29:
If using Band 1, 2.4 GHz, you probably don't need the following lines active:

ieee80211ac=1
ieee80211ac=1

The following line doesn't need to be active unless you need outdated security methods:

wpa_pairwise=TKIP

I use the following for my rtl8812au based adapter:

ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]

hostapd will not work if you try to use a capability that the adapter does not have. You can check the Band 1 capabilities as follows:

$ iw list

You are obviously try to set up an AP but it isn't clear to me where the internet access is coming from and how you have networking set up to handle it. You say you don't have a br0 interface. What is the status at this point?

CpServiceSpb, 2021-08-01:
I have quite complicated configuration.
I have Freeradius as Dhcp server and Bind9 with a different answers regarding on request IP (views) as Dns one.
I won Wi-fi - RTL8812au as access point now at 2.4 GHz only, 5 GHz did not test yet.
The message I write through A Windows client connected via Intel Centrino Wireless 6150-N Wi-fi card to Alfa AWUS036AC (RTL8812au) plugged to USB3 port of Supermicro server booted up to Ubuntu 18.04 !
There is 2 questions which requires answers:
1. Why is 54 mbit/s at Wi-fi connection property on Windows, though as card as AP supports faster speed;
2. Speedtest shows up two 2 times less real speed, up to 21 mbit/s. I measured the speed to ISP site;
3. Now Freeradius run from root. It is necessary to turn its running to other user, but there is some break point.

morrownr, 2021-08-01:
Whew, that is complicated. Let me answer what I can:

1. I do not fully understand the question. Can you rework the question? 

2. The speed you get with iperf3 or other testing software will always be half to slightly less than half of the indicated Mb/s of the connection.

3. I am not familiar with Freeradius so I am not the person to ask.

4. Realtek drivers don't support this. The Mediatek/Ralink drivers that are in-kernel do:

More info about usb wifi adapters that support multi SSIDs and use in-kernel drivers:

https://github.com/morrownr/USB-WiFi

Here is a link to a specific adapter that supports up to 8 SSIDs on the same freq:

https://store.rokland.com/collections/wi-fi-usb-adapters/products/alfa-awus036neh-802-11n-wireless-n-usb-wi-fi-adapter

CpServiceSpb, 2021-08-01:
1. Windows 7 wi-fi cinnection property shows only 54 mbit/s.
2. Why ?
Both for point 1 and 2 tells (for me) that the client connected at 54 mbit/s. I don' t understand why.
3. I simply shared the information. Anyway I will work on this.
4. Is it realtek driver restriction/misfunctioning or firmware or 8812au chip in its own ?

morrownr, 2021-08-01:
1. I understand the question now. The answer is: I don't know. My history with operating systems on small computers is CP/M, PCDOS, DRDOS, OS/2 and Linux. I simply don't know much about Windows and have no capability at all to test it.

2. half-duplex. If your system shows a link speed of 867 Mb/s then under perfect conditions you can expect a measured speed of around 400 Mb/s.

4. You can see what the reported capabilities are by running:

$ iw list

With Realtek drivers, you will get:

"interface combinations are not supported"

With the Mediatek/Ralink in-kernel drivers, it will tell you what is supported and it can vary from adapter to adapter. Examples:

The Alfa ACM reports the following:

 valid interface combinations:
                 * #{ IBSS } <= 1, #{ managed, AP, mesh point, P2P-client, P2P-GO } <= 2,
                   total <= 2, #channels <= 1, STA/AP BI must match

That means the Alfa ACM can be run with various combination such managed and AP interfaces at the same time or two AP interfaces.

The Alfa NEH reports the following:

Valid interface combinations:
		 * #{ AP, mesh point } <= 8, total <= 8, #channels <= 1

That means you can run up to 8 AP interfaces or a combination of APs and mesh.

To answer why the Realtek driver does not support interface combinations, you would have to ask Realtek... but good luck with that because I have found no way to communicate with Realtek and I have tried.

CpServiceSpb, 2021-08-02:
What is signal signal stronger, more stable - at AWUS036AC or at AWUS036ACM ?

morrownr, 2021-08-02:
I do not have an Alfa AC so I cannot do a direct comparison. I do have an Alfa ACH and an Alfa ACM and I have tested then against each other. For me, the winner for AP mode is the ACM. It is incredibly stable and the signal is strong enough for what I do. I have compared it to the 5 GHz output on my Zyxel Armor Z2 AC2600 Wireless Router and the signal strength of the ACM and Zyxel router are basically equal. For me, the winner in client mode is the ACH as it seems to have a little longer range in client mode.

There is one BIG difference though. The Alfa ACH and AC both depend on out-of-kernel drivers. That is problematic at best. It has been nearly a year since the last out-of-kernel code was made available from a seller. The 8812au chipset has been being replaced by the 8812bu and we don't know if Realtek will ever release another updated version of the 8812au driver. This 8812au driver has gaps in capability such as lack of WPA3 capability.

On the other hand, the driver for the Alfa ACM is in the Linux kernel. This means several things:

- You don't have to install a driver. Plug the adapter in and it works.
- You don't don't have to worry about a kernel upgrade breaking the driver.
- You don't have to worry about new versions of the driver being released as once a driver is in-kernel, you are probably looking at it being supported for 25+ years or more.
- You have a place to report bugs: https://wireless.wiki.kernel.org/en/users/documentation/reporting_bugs
- You can submit also submit code that fixes bugs or adds capability.
- In-kernel drivers are Linux Wireless Standards compatible (Realtek drivers are not)
- The in-kernel driver for the Alfa ACM already supports WPA3 (Realtek ? who knows when ?)

My point is that there is more to picking an adapter than what most folks realize. Don't underestimate how good it is to have a good quality adapter that uses an in-kernel driver. The Alfa ACM is a good quality adapter is its in-kernel driver is good. In my opinion, it is the best wifi adapter that Linux users can buy. If I were limited to only owning one wifi adapter, the Alfa ACM would be it.

In the end, I suspect the Alfa AC and ACM have similar signal strength characteristics but for Linux users, there is a really big difference when it comes to driver support, making the ACM the clear winner.

CpServiceSpb, 2021-08-02:
Thanks for expanded response.

CpServiceSpb, 2021-08-03:
Finally I ordered AWUS036ACM.

morrownr, 2021-09-27:
How about a report?

CpServiceSpb, 2021-09-27:
We moved to separate external access point with 2 radio simultaneously on
2.4 GHz and 5 Ghz.

Arty.

morrownr, 2021-09-28:
FYI: We are working on a new version of this driver. It is based on very fresh source and should be an improvement. I will post notice on the repo here in the README.

CpServiceSpb, 2021-09-28:
Hi. I will wait to look at it.

Regards ...

closed 2021-11-24 by morrownr