morrownr/8812au #34: Is there planning to add make dkms_install ?

CpServiceSpb, 2021-07-26:
There is make dkms_install / make dkms_delete as make target at https://github.com/aircrack-ng.
Are there plans to do so here ?

morrownr, 2021-07-26:
Have you read the installation instructions?

morrownr, 2021-07-27:
Disregard my previous reply. I had taken your comment as asking if dkms is used but I now see that is not what you were asking. My bad.

To answer the question: I try not to rule anything out. I have taken a short look at the work you are pointing out. I'm not sure that going that direction is consistent with the objectives of this repo.

My primary objective here is to help increase the number of Linux users. The vast majority of computer users do not know what "make" is and aren't really interested in learning. They simply use a computer to get a job done. Making Linux "just work" will help computer users that try Linux, stay on Linux. A Linux user that wants to set up a computer for grandma that lives in another city has to know that the computer will just work and will continue to work long term.

These Realtek drivers, being out-of-kernel, are not consistent with having long term stable, reliable systems so I support these drivers for Linux users that have Realtek adapters but I make the information available regarding in-kernel supported adapters so that Linux users can have long term stable systems:

https://github.com/morrownr/USB-WiFi

For those that want to work on driver code, the opportunity is there.  Work on these drivers:

https://github.com/torvalds/linux/tree/master/drivers/net/wireless/mediatek/mt76

You can submit fixes and report bugs for the Mediatek drivers:

https://wireless.wiki.kernel.org/en/users/documentation/reporting_bugs

There is no way, that I am aware of, to submit bug reports or fixes to Realtek. I have talked about this to retailers and I quote "it seems Realtek has just thrown up their hands" and said retailer is not just talking about Linux support but across the board.So my question is "if Realtek isn't providing support and doesn't seem to want to do their drivers the way, why should we use their products?"

I plan to continue trying to keep these Reltek drivers going but I strongly urge Linux users that use usb wifi adapters to look at using adapters with in-kernel drivers.

CpServiceSpb, 2021-08-01:
I made some modifiction of Makefile, dkms.conf being taken from other 8812au github pproject.
So, as make dkms_install as dkms_remove targets worked well at the time.
I can send it to inclulde ti this project.

morrownr, 2021-08-01:
If you send your work, I will take a look. I can't promise I will use it but I am always on the lookout for things that will make things better.

CpServiceSpb, 2021-08-03:
Tell e-mail where I should send makefile patch and dkms.conf for supporting of make dkms_install/dkms_remove.

morrownr, 2021-09-28:
I have been very busy but have not forgot this issue. Currently I am working to make a new version of this driver available. Once that has happened, please contact me and I will look at how we might slide this capability in there.

Nick

closed 2021-11-24 by morrownr

CpServiceSpb, 2021-11-28:
Hi.
How is your way of reaching of new drivers ?
Is it easy or hard ?

morrownr, 2021-11-29:
Hi @CpServiceSpb 

The new version of the driver is located...

https://github.com/morrownr/8812au-20210629