morrownr/8812au #35: Alfa ‎AWUS036ACH kernel error

BitVenturesUSA, 2021-08-01:
Hello, I've been getting this error for quite a while with this and other 8812au drivers. The device is an Alfa ‎AWUS036ACH. Performance slowly degrades until I unplug it and plug it back in. I have attached some system information below, please let me know if you need anything further. Thanks!

```
[178845.644804] WARNING: CPU: 1 PID: 987 at /var/lib/dkms/rtl8812au/5.9.3.2/build/hal/hal_com_phycfg.c:2252 phy_chk_ch_setting_consistency+0x71/0xe5 [8812au]
[178845.644805] Modules linked in: cdc_ether(E) usbnet(E) r8152(E) mii(E) md4(E) sha512_ssse3(E) sha512_generic(E) cmac(E) nls_utf8(E) cifs(E) libarc4(E) dns_resolver(E) fscache(E) libdes(E) udp_diag(E) tcp_diag(E) inet_diag(E) nfnetlink_queue(E) xt_NFQUEUE(E) xfrm_user(E) xfrm_algo(E) binfmt_misc(E) nls_cp437(E) vfat(E) fat(E) nf_log_ipv6(E) ip6t_REJECT(E) nf_reject_ipv6(E) intel_rapl_msr(E) intel_rapl_common(E) xt_hl(E) ip6_tables(E) ip6t_rt(E) 8812au(OE) nf_log_ipv4(E) x86_pkg_temp_thermal(E) nf_log_common(E) cfg80211(E) ipt_REJECT(E) intel_powerclamp(E) nf_reject_ipv4(E) coretemp(E) kvm_intel(E) xt_LOG(E) nft_limit(E) evdev(E) kvm(E) irqbypass(E) xt_limit(E) iTCO_wdt(E) xt_addrtype(E) intel_pmc_bxt(E) ee1004(E) rapl(E) intel_cstate(E) tpm_infineon(E) hp_wmi(E) iTCO_vendor_support(E) wmi_bmof(E) sparse_keymap(E) intel_uncore(E) xt_tcpudp(E) rfkill(E) watchdog(E) sg(E) tpm_tis(E) tpm_tis_core(E) tpm(E) efi_pstore(E) intel_pmc_core(E) intel_pch_thermal(E) pcspkr(E) rng_core(E) acpi_pad(E)
[178845.644841]  xt_conntrack(E) nf_conntrack(E) nf_defrag_ipv6(E) nf_defrag_ipv4(E) nft_compat(E) nft_counter(E) nf_tables(E) nfnetlink(E) parport_pc(E) ppdev(E) lp(E) fuse(E) parport(E) configfs(E) efivarfs(E) ip_tables(E) x_tables(E) autofs4(E) dm_crypt(E) dm_mod(E) raid10(E) raid456(E) async_raid6_recov(E) async_memcpy(E) async_pq(E) async_xor(E) async_tx(E) raid1(E) raid0(E) multipath(E) linear(E) md_mod(E) xfs(E) btrfs(E) blake2b_generic(E) xor(E) raid6_pq(E) libcrc32c(E) ext4(E) crc16(E) mbcache(E) jbd2(E) crc32c_generic(E) nls_ascii(E) sd_mod(E) t10_pi(E) crc_t10dif(E) crct10dif_generic(E) crct10dif_pclmul(E) crct10dif_common(E) crc32_pclmul(E) crc32c_intel(E) ghash_clmulni_intel(E) i915(E) i2c_algo_bit(E) drm_kms_helper(E) cec(E) ahci(E) e1000e(E) ptp(E) xhci_pci(E) libahci(E) pps_core(E) xhci_hcd(E) libata(E) usbcore(E) usb_common(E) aesni_intel(E) libaes(E) crypto_simd(E) scsi_mod(E) cryptd(E) glue_helper(E) i2c_i801(E) i2c_smbus(E) drm(E) wmi(E) video(E) button(E)
[178845.644882] CPU: 1 PID: 987 Comm: RTW_CMD_THREAD Tainted: G        W  OE     5.10.0-8parrot1-amd64 #1 Debian 5.10.46-8parrot1
[178845.644883] Hardware name: HP HP EliteDesk 800 G2 DM 35W/8055, BIOS N21 Ver. 02.53 04/16/2021
[178845.644916] RIP: 0010:phy_chk_ch_setting_consistency+0x71/0xe5 [8812au]
[178845.644918] Code: 2d 0f 0b 83 3d 44 53 13 00 02 77 59 b8 00 00 00 00 48 8b 54 24 08 65 48 2b 14 25 28 00 00 00 75 78 48 83 c4 10 5b 5d 41 5c c3 <0f> 0b eb d5 80 fb 0e 77 19 83 7d 44 00 75 0f b8 01 00 00 00 83 7d
[178845.644919] RSP: 0000:ffffa58c408a7d18 EFLAGS: 00010246
[178845.644921] RAX: 0000000000000000 RBX: 00000000000000a3 RCX: ffffa58c4058a2b6
[178845.644921] RDX: 00000000000000a3 RSI: ffffa58c408a7d1f RDI: 00000000000000a2
[178845.644922] RBP: ffffa58c40585000 R08: 000000000000007d R09: 0000000000000000
[178845.644923] R10: 0000000000000001 R11: 0000000000000000 R12: 00000000000000a3
[178845.644923] R13: 00000000000000a3 R14: 000000000000009b R15: 0000000000000000
[178845.644925] FS:  0000000000000000(0000) GS:ffff8fd61f880000(0000) knlGS:0000000000000000
[178845.644925] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[178845.644926] CR2: 00007f80102c2f08 CR3: 0000000101efa002 CR4: 00000000003706e0
[178845.644927] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[178845.644928] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[178845.644928] Call Trace:
[178845.644962]  ? rtw_hal_write_rfreg+0x16/0x17 [8812au]
[178845.644994]  rtw_hal_set_tx_power_level+0x1e/0x52 [8812au]
[178845.645024]  phy_SwChnlAndSetBwMode8812+0x7c/0x143 [8812au]
[178845.645053]  PHY_HandleSwChnlAndSetBW8812+0x1b3/0x1b8 [8812au]
[178845.645082]  PHY_SetSwChnlBWMode8812+0x29/0x2e [8812au]
[178845.645112]  rtw_hal_set_chnl_bw+0xcb/0x14f [8812au]
[178845.645134]  set_channel_bwmode+0x2d7/0x2e1 [8812au]
[178845.645153]  ? rtw_set_external_auth_status+0xfa/0xfa [8812au]
[178845.645172]  rtw_set_chbw_hdl+0x4e/0x94 [8812au]
[178845.645184]  rtw_cmd_thread+0x473/0x65c [8812au]
[178845.645196]  ? rtw_stop_cmd_thread+0x39/0x39 [8812au]
[178845.645199]  kthread+0x11b/0x140
[178845.645201]  ? __kthread_bind_mask+0x60/0x60
[178845.645203]  ret_from_fork+0x22/0x30
[178845.645205] ---[ end trace f7d7a58ad4eabb2a ]---
```

```
`Bus 002 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         9
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x8812 RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
  bcdDevice            0.00
  iManufacturer           1 Realtek
  iProduct                2 802.11n NIC
  iSerial                 3 123456
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0053
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              800mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
[... 72 lines cut ...]
    bDevCapabilityType      2
    bmAttributes   0x00000002
      HIRD Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat         255 micro seconds
Device Status:     0x0004
  (Bus Powered)
  U1 Enabled

```
```
Linux [hostname] 5.10.0-8parrot1-amd64 #1 SMP Debian 5.10.46-8parrot1 (2021-07-21) x86_64 GNU/Linux
`
```

morrownr, 2021-08-01:
Hi, welcome.

I have an Alfa ACH and will set it up to see what I can find out.

Please post the options line from /etc/modprobe.d/8812au.conf

Tell me about your USB ports.

Have you tried more than one USB port? If so, same results?

Have you tested with an OS besides Parrot? Such as Kali?

morrownr, 2021-08-01:
I've been working on a system with the Alfa ACH for a couple of hours now and am not seeing any problems. Log are clean. Throughput ranges from 400 Mb/s to 200 Mb/s depending on which 5 GHz AP I am using.

The test system is Dell Vostro with i7 processor and Linux Mint 20.2 with kernel 5.4. I recently moved and don't have my lab set up yet but I do have one additional system that I can test with and it has Kali, Manjaro and Raspberry Pi OS (x86 edition.) 

I will wait for your answers to the above questions before changing the ACM to another system.

BitVenturesUSA, 2021-08-02:
> Hi, welcome.
> 
> I have an Alfa ACH and will set it up to see what I can find out.
> 
> Please post the options line from /etc/modprobe.d/8812au.conf
`options 8812au rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_switch_usb_mode=1 rtw_beamform_cap=0`

> 
> Tell me about your USB ports.
This is the internal USB3.0 hub it's plugged into. Rear ports. 
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         3
  bMaxPacketSize0         9
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0003 3.0 root hub
  bcdDevice            5.10
  iManufacturer           3 Linux 5.10.0-8parrot1-amd64 xhci-hcd
  iProduct                2 xHCI Host Controller
  iSerial                 1 0000:00:14.0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
        bMaxBurst               0
Hub Descriptor:
  bLength              12
  bDescriptorType      42
  nNbrPorts            10
  wHubCharacteristic 0x000a
    No power switching (usb 1.0)
    Per-port overcurrent protection
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  bHubDecLat          0.0 micro seconds
  wHubDelay             0 nano seconds
  DeviceRemovable    0x00 0x00
 Hub Port Status:
   Port 1: 0000.0223 5Gbps power U1 enable connect
   Port 2: 0000.02a0 5Gbps power Rx.Detect
   Port 3: 0000.02a0 5Gbps power Rx.Detect
   Port 4: 0000.02a0 5Gbps power Rx.Detect
   Port 5: 0000.02a0 5Gbps power Rx.Detect
   Port 6: 0000.02a0 5Gbps power Rx.Detect
   Port 7: 0000.02a0 5Gbps power Rx.Detect
   Port 8: 0000.02a0 5Gbps power Rx.Detect
   Port 9: 0000.02a0 5Gbps power Rx.Detect
   Port 10: 0000.02a0 5Gbps power Rx.Detect
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x000f
  bNumDeviceCaps          1
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x02
      Latency Tolerance Messages (LTM) Supported
    wSpeedsSupported   0x0008
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   3
      Lowest fully-functional device speed is SuperSpeed (5Gbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat         512 micro seconds
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

> 
> Have you tried more than one USB port? If so, same results?
Yes, I've tried all available USB3.0 ports except the ones designated for KB/Mouse and same result. 
> 
> Have you tested with an OS besides Parrot? Such as Kali?
I know I had similar issues with Kali many months ago, but I can't say for certain I was using your driver and/or if it was the same exact issue. Wish I had more data for you.

I dont know if it helps, but here is the sequence I use to get this puppy going:
```
ip link set dev wlan0 down
macchanger -A wlan0
airmon-ng check kill [depreciating this soon as I've managed to exclude wlan0 from NetworkManager's greasy hands]
iw reg set US
iw wlan0 set monitor control
iw dev wlan0 set txpower auto
ip link set dev wlan0 up
```

morrownr, 2021-08-02:
Try this to keep NetworkManager's greasy hands off:

Ensure Network Manager doesn't cause problems
```
sudo nano /etc/NetworkManager/NetworkManager.conf
```
add
```
[keyfile]
unmanaged-devices=interface-name:wlan0;interface-name:mon0
```
Note: The above tells Network Manager to leave the wlan0 and mon0 interfaces
alone.

Now back to our regularly scheduled programming:

Suggest trying this:

options 8812au rtw_drv_log_level=4 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=0 rtw_switch_usb_mode=2 rtw_beamform_cap=10

I made several changes. Let's get some driver log entries going. Check dmesg. Let's turn power management off. Let's go with USB2 for now. And let's go with the regular default for beamforming. If you see good results then you can put one setting at a time back in place to see if we can figure out which was the problem.

Clarify something for me? This is for client mode and monitor mode? No AP mode?

Also, hopefully I can move the ACH over to my system with Kali and kernel 5.10 later this week.

closed 2021-11-24 by morrownr