morrownr/8812au #36: Possible to force 11g in 2.4 GHz band?

antonott, 2021-08-02:
Hi, and thank you for your excellent work! After several days of struggle I finally went out and bought a wifi adapter with the RTL8812AU chipset, and with the help of your drivers I finally managed to do packet injection with [Scapy](https://scapy.net/).

I still have one issue however. My use case is that I'm doing an experiment with SDR (Software Defined Radio), and I need a source of 802.11 frames that my SDR can receive and decode. I'm using [GNU Radio](https://www.gnuradio.org/) and [a module](https://github.com/bastibl/gr-ieee802-11) from the [Wime Project](https://www.wime-project.net/) for the SDR part, and an ASUS USB-AC56 (with RTL8812AU chipset) to inject the frames with the help of Scapy. When I inject in the 5+ GHz band this works great, but when I inject in the 2.4 GHz band I can't seem to capture any frames with the SDR.

Based on what is written [here](https://www.wime-project.net/tutorials/wifi-rx/), I suspect that my wifi adapter is using 802.11b compatability mode, which my SDR cannot decode. I need it to use 802.11g. `iwlist <interface> modulation` returns only `unknown modulation information`, and when trying things like `sudo iwconfig <interface> modu 11g` I get `Operation not supported` or `invalid argument`.

So, is there a way to force 11g in the 2.4 GHz band?

morrownr, 2021-08-02:
I understand. Let me see. I'll take a look around.

morrownr, 2021-08-02:
Try this:

add the following to the "options" line in /etc/modprobe.d/8812au.conf

rtw_wmm_enable=0

That is an indirect way of forcing 802.11g because wmm is required for 802.11n.

Maybe that will do what you want. Let me know.

antonott, 2021-08-03:
Thank you for the suggestion, but unfortunately it doesn't seem to do the trick. I still see nothing on the air with my SDR when transmitting in the 2.4 GHz band. (This doesn't mean nothing is on the air, just that there is nothing that the SDR can decode.)

antonott, 2021-09-06:
Sorry for not responding sooner! My project ended a couple of weeks ago, so I haven't had access to the USRP devices in a little while. I still intend to try your suggestion at some point though, and I will get back to you then with the results.

morrownr, 2021-09-06:
No problem. I know how it goes. I would be interested in knowing if that accomplished what you were looking for.

closed 2021-11-24 by morrownr