morrownr/8812au #39: Error in dkms_remove

jcollum, 2021-09-06:
```
$ sudo make dkms_remove
dkms remove 8812au/4.2.3 --all
Error! There are no instances of module: 8812au
4.2.3 located in the DKMS tree.
```

Had to run `sudo dkms remove rtl8812au/5.9.3.2 --all` to get it to work. 

BTW installing this in `Linux Ubuntu 5.8.0-43-generic #49-20.04.1-Ubuntu` caused some issues. I've been fussing around with wifi drivers. When I installed this one (at the suggestion of someone a good deal more experienced than me) my machine became unbootable. You can see the full conversation here: https://chat.stackexchange.com/rooms/129273/discussion-between-chili555-and-jcollum

morrownr, 2021-09-06:
Hello @jcollum 

I know chili555. We chat periodically. He knows his stuff. I took a look at the discussion.

Now, how we proceed is up to you. I can only offer some suggestions.

First, there is no "error in dkms_remove" with this driver because there is no such thing as "dkms_remove" with this driver.

You appear to be mixing the instructions from another driver with this driver. When it comes to Realtek out-of-kernel drivers, there is little standardization regarding how devs will present the driver for use. You have to follow the instructions for the driver that you decide to use. The instructions here guide you to use "remove-driver.sh".

Before I go on with some options as to how get you back in business, let me provide some information. I've been using Linux since around '94. It has grown up into a very nice operating system but it can take those that are new to the os a little while to get used to it. One of the reasons I established this site is to provide as much help for those that are new that have Realtek usb adapters. This site has drivers that cover 9 Realtek usb chipsets and all drivers are installed, used and removed in the same way. I've tried to make it as easy and user friendly as possible.  There is only so much that I can do as Realtek is not doing its Linux usb adapter support the right way. Out-of-kernel drivers are problematic on Linux. Therefore, before we go on, I am going to provide you with a link that should provide a lot of good information:

https://github.com/morrownr/USB-WiFi

Please take the time to read that site all the way through. There is a section on Realtek drivers at the end. Once you are complete, you will know a lot about usb wifi adapters on Linux.

Okay, with that said, you very well may have a very messed up system. Do you want to do a "format c:" followed by a fresh installation or do you want help trying to figure this out with the current installation?

jcollum, 2021-09-06:
Well in the meantime I decided to install Linux mint instead. Didn't really
have any better luck getting it to see the Wi-Fi. But for an HTPC server I
prefer Mint to Ubuntu.

morrownr, 2021-09-06:
I prefer Mint also. Gnome just doesn't do anything for me.

Now, if you installed the current Mint and followed the instructions to install this driver as shown in the README, it should work... unless your adapter does not have a 8812au chipset.

What does the following show?

$ lsusb

jcollum, 2021-09-06:
I'm working through a possible hard disk failure right now -- having all
kinds of issues with this system! I'll get back to you but it might be a
while and I can't guarantee I will remember.

morrownr, 2021-09-07:
Good luck on getting things going. Come back when able.