morrownr/8812au #40: Using hostapd, clients cannot connect to each other on AP lan

Mario100000, 2021-09-07:
First of all, thanks for this git driver release. It also compiles with the latest version of linux 5.14.0.

I use it with hostapd. It works fine but there seems there to be an issue well described here:
https://unix.stackexchange.com/questions/645513/clients-cannot-connect-to-each-other-on-wlan-no-reply-on-arp-requests

I don't know which driver are used in the linked message.
I have the same issue using the module 8812au compiled by this git.

Is the issue (clients cannot to connect to each other) due to the driver?

morrownr, 2021-09-07:
> First of all, thanks for this git driver release. It also compiles with the latest version of linux 5.14.0.

You are welcome and thanks for letting me know that it is compiling properly on 5.14. I suspected it would work with 5.14 as I had not seen anything during the 5.14 dev cycle that would require updates but it is very good to get confirmation as I have not had time to test.
 
> I use it with hostapd. It works fine but there seems there to be an issue well described here:
> https://unix.stackexchange.com/questions/645513/clients-cannot-connect-to-each-other-on-wlan-no-reply-on-arp-requests

Hmmm...

> I don't know which driver are used in the linked message.
> I have the same issue using the module 8812au compiled by this git.
> 
> Is the issue (clients cannot to connect to each other) due to the driver?

It very well could be due to this driver. This is a very heavily used driver but seems to mostly be used for managed and monitor modes. I test it with hostapd but can't say that I have tried what you are doing as I just do short test as it is not my regular operational setup. I am a little behind with some other issues I am working so it could be a while before I could take a look at this... and I don't know if I can fix it when I do look.

In the meantime, I'll throw a little info out for you to think about:

My hostapd setup runs on a RasPi4b with internal wifi turned off. I use 2 usb wifi adapters so as to have dual band. 5 GHz is handled by an ALFA AWUS036ACM and 2.4 GHz is handled by an ALFA AWUS036ACHM. Both adapters use in-kernel drivers and wow, does this work well.

A guide to my setup can be found in the document Bridged Wireless Access Point found in this repo:

https://github.com/morrownr/7612u

I also maintain a repo regarding usb wifi adapters just for Linux users:

https://github.com/morrownr/USB-WiFi

I just tested your issue with my setup and it works well. I tested both bands on the AP so the driver for the mt7612u chipset and the driver for the mt7610u chipset are both working well in this regard.

Keep me posted.

Mario100000, 2021-09-08:
It's a pity that manufacturer like Realtek don't support their product in Linux mainline!

I agree with your written (https://github.com/morrownr/USB-WiFi):
The rtl8812au chipset is an aging chipset that will likely never be supported by an in-kernel driver which leaves users dependent on Realtek to release future out-of-kernel source code.
...
The above driver is a good quality driver but the question is, how long will Realtek continue to release out-of-kernel source code for this chipset.

I knew atheros devices have good in-kernel support (I'm using some M.2 boards with Qualcomm Atheros QCA6174). In USB I used rtl8192cu adapter and now rtl8812au one. I haven't considered Mediatek devices, I'll do it now.

I confirm my AP configuration works fine with QCA6174 (connection between station is possible).

Thanks.

morrownr, 2021-09-08:
> It's a pity that manufacturer like Realtek don't support their product in Linux mainline!

This is true. It is not clear to me why Realtek has made the decisions they have made. What I do know is that most Linux users that use or want to use usb wifi adapters would be happier using adapters based on Mediatek chipsets. 

> I agree with your written (https://github.com/morrownr/USB-WiFi):
> The rtl8812au chipset is an aging chipset that will likely never be supported by an in-kernel driver which leaves users dependent on Realtek to release future out-of-kernel source code.

I understand that good information has been lacking. One thing I try to do in that document is increase awareness among Linux users that they need to be careful if they are looking to make a purchase. The in-kernel Mediatek drivers will likely be supported and improved in the Linux kernel for the next 20 years and probably longer. On the other hand, we may never see another release of a driver from Realtek. The 8814au chipset has not seen a driver release since 2019 and that was not a good driver. In fact, that driver and the one for the 8811au are getting hard to maintain. I hope to keep both going through another LTS kernel release but if we don't see an update after the next LTS release, I may look at putting further maintenance on hold.

> The above driver is a good quality driver but the question is, how long will Realtek continue to release out-of-kernel source code for this chipset.

I say "good  quality driver" but I should say "good quality driver, as far as Realtek drivers are concerned." 

> I knew atheros devices have good in-kernel support (I'm using some M.2 boards with Qualcomm Atheros QCA6174). In USB I used rtl8192cu adapter and now rtl8812au one. I haven't considered Mediatek devices, I'll do it now.

Atheros, Intel and Mediatek all have good in-kernel wifi support. Since Intel and Atheros are not providing usb chipsets to the market, we are basically just left with Mediatek for usb adapters. However, as you can see in the USB-WiFi document, there are a lot of Mediatek based adapters available and I am seeing more coming on the market. There is already a usb wifi driver in the kernel for the mt7663u, which appears to be a Wave 2 chipset. I'm not seeing adapters with this chipset on the market yet.
 
> I confirm my AP configuration works fine with QCA6174 (connection between station is possible).

Good.