morrownr/8812au #41: Error install

jur3ma, 2021-09-12:
Your kernel headers for kernel 5.10.63-v7+ cannot be found at
/lib/modules/5.10.63-v7+/build or /lib/modules/5.10.63-v7+/source.
An error occurred. dkms build error = 1
Please report this error.

morrownr, 2021-09-13:
What OS and version are we talking about? What hardware?

jur3ma, 2021-09-13:
Raspberry OS Lite Buster 2021-05-07, Raspberry Pi 4..

morrownr, 2021-09-14:
Did you follow the installation intructions in README?

Did you run the following line?

$ sudo apt install -y raspberrypi-kernel-headers bc build-essential dkms git

jur3ma, 2021-09-14:
Yes, I followed the indicated instructions to the letter, it is as if that folder did not exist in the Kernel

morrownr, 2021-09-14:
It appears I am going to have to burn a sd with RasPi OS Lite and see what is going on. I use the other two editions of RasPiOS without issues so something must be different about the Lite edition. It may take me a couple of days, possible more, to knock this out. While I am working on it from this end, can you search to see if anyone else is seeing something like this with the Lite edition?

jur3ma, 2021-09-14:
I don't know, I'm going to try Dietpi to see if the problem persists, if not, I use the RasPi OS  Desktop

jur3ma, 2021-09-14:
dietpi para mi es muy engorroso, paso a probar la RasPi OS Desktop

morrownr, 2021-09-14:
Yo entiendo.

jur3ma, 2021-09-14:
same mistake

DKMS: add completed.
Error! echo
Your kernel headers for kernel 5.10.17-v7+ cannot be found at
/lib/modules/5.10.17-v7+/build or /lib/modules/5.10.17-v7+/source.
An error occurred. dkms build error = 1
Please report this error.

morrownr, 2021-09-14:
When I run the following:

$ uname -r

I get: 5.10.60-v7l+

Why is your kernel so much older than mine? Try this:

$ sudo apt update
$ sudo apt full-upgrade

Then run the full installation checklist again and see what happens?

jur3ma, 2021-09-15:
big mistake on my part, at no time reboot after updating, that's why the new kernel did not load, this time the installation was done well, but the USB wifi does not raise me, although doing an lsusb appears, but when doing a ifconfig only shows the ethernet and wifi of the pi, but not the USB

jur3ma, 2021-09-15:
my intention is to put the pi 3 as a wifi repeater, that is, deactivate the wifi of my router and manage it on the Pi 3

morrownr, 2021-09-15:
Quote: "ID 0bda:a811 Realtek Semiconductor Corp. RTL8811AU 802.11a/b/g/n/ac WLAN Adapter"

jur3ma, you are installing the wrong driver. Your adapter is based on the 8811au chipset, not the 8812au.

The driver you need is... 8821au

Go out to the main site:

https://github.com/morrownr

You will see the repo for the 8821au.

You might also be interested in reading USB-WiFi.

You can run remove-driver.sh for this 8812au driver and then reboot if you want to remove it.

Let me know how it goes.

jur3ma, 2021-09-15:
well, now if it has been installed and appears in ifconfig, now I have to make it a wifi point, thank you very much and sorry for the inconvenience and my ignorance.

morrownr, 2021-09-15:
There is no need to apologize. I make my share of mistakes.

In fact, at one time the 8812au driver did support the 8811au chipset so you might see references to that on the internet. They were broken into separate drivers about 2 years ago.

I see you want to use the adapter on a RasPi and set up an access point. I have extensive experience along those lines. If you want to start an issue over in the 8821au repo with a title like "Using a RasPi as an access point" I will be glad to provide you with a lot of good information.

Que tenga un buen dia.

Nick