morrownr/8812au #42: wifi usb supports AP mode

jur3ma, 2021-09-15:
How to know if anten wifi usb supports AP mode?

Bus 001 Device 004: ID 0bda:a811 Realtek Semiconductor Corp. RTL8811AU 802.11a/b/g/n/ac WLAN Adapter

THX

morrownr, 2021-09-15:
Quote: "How to know if anten wifi usb supports AP mode?"

Run the follow command:

$ iw list

You will likely see the output from both your usb adapter and the internal wifi chipset. It is easy to tell the difference in this case because the RasPi3B only supports Band 1 (2.4g). You need to look for the section similar to the below:

	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
		 * P2P-client
		 * P2P-GO

See where it says "AP"? If it says AP then it supports master mode (AP) and you are good to use it as an access point.

jur3ma, 2021-09-15:
Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * monitor
                 * P2P-client
                 * P2P-GO
is AP?

morrownr, 2021-09-15:
Yes

Do you have a plan for how to put the adapter into AP mode? I ask because this is not so easy if you have no experience doing it.

jur3ma, 2021-09-15:
ready, I have been able to configure it following the tutorial that is in the driver folder, it works correctly, now I want to install a pihole and dns service

jur3ma, 2021-09-15:
can you have the AP, pihole and a dns server on the same pi?

morrownr, 2021-09-15:
> can you have the AP, pihole and a dns server on the same pi?

Yes but I am not saying it is a good idea or bad because I don't fully know what your goal is.

jur3ma, 2021-09-16:
My goal is to have a connection as clean as possible, hence having a network point that I know who administers it, and that serves as a filter for my router, that's why the pihole, the vpn server is to have no restrictions to navigate and watch channels from other countries.

morrownr, 2021-09-16:
It sort of sounds to me like you want a "routed wireless access point" instead of a "bridged wireless access point." I hope to add a guide for setting up a routed wireless access point at some point but I do not have time right now.  Take a look at this:

https://docs.raspap.com/

Other news: I located an updated version of the 8821au driver today (and here we are talking about it in the 8812au driver repo). It is a dramatically updated version that includes support for WPA3. Hopefully I can make it available soon but you need to hang out over in the 8821au repo to get the news and updates since that is the driver you are actually using.

jur3ma, 2021-09-22:
It would be fantastic if you made a guide, I followed the one you have, the first time, but yes, I want the access point piped, I will be attentive to you update the driver (if you have not done it yet) since you create the access point guide routed. Thanks a lot for everything, kind regards!!

jur3ma, 2021-09-23:
I have followed the steps in the guide for the routed access point, at the moment I have tried the fast way (automatic), everything is installed, the network is raised, but it does not assign ip ...... I will try the manual method ....

morrownr, 2021-09-24:
Since your Device ID indicates your chipset is a rtl8811au, let me point out that a new driver for this chipset is now available at:

https://github.com/morrownr/8821au-20210708

This is a very new version of the driver. Testing is still ongoing but everything appears to be working well. AP mode appears to function better on this new driver than on any Realtek driver I have ever used.