morrownr/8812au #44: (discussion) disable hardware encryption possible?

Matheus-Garbelini, 2021-09-26:
Hi, I was wondering if it's possible to disable hardware encryption for this driver and enable only software encryption
as referenced here on the source code: https://github.com/morrownr/8812au/blob/5.9.3.2/core/rtw_recv.c#L645
This function is never executed because `prxattrib->bdecrypted` is always true when we receive a frame.

@morrownr do you know if it's possible to modify something in the source code to modify this behaviour?

Regards.

morrownr, 2021-09-26:
> @morrownr do you know if it's possible to modify something in the source code to modify this behavior?

@Matheus-Garbelini 

I do not know what is possible regarding this issue but am will to explore it with you, however, here is a status report to let you know it may take a few days to get to it:

I was very recently able to obtain new clean source for the 8812au and 8821au (supports the 8811au and 8821au chipsets). Given that the existing 8821au driver is not that good, I prioritized bringing the 8821au online first so have been putting most of my effort there. Tests so far on the [new 8821au driver](https://github.com/morrownr/8821au-20210708) are showing it to be a really big improvement over the old version. I hope to have things wrapped up with the new 8821au driver at some point this week so I can start pointing people to the new version...at which time I will concentrate of bringing the new 8812au driver to life. It is v5.13.6 (20210629). After seeing your message, I did a quick comparison of the /core/rtw_recv.c files in the current and new versions and the new version has a VERY LARGE amount of additions and changes. Therefore, I think it is best if we wait on this issue until after I can get the new version up.

I will need an extra tester or 2. Are you able to test?

Matheus-Garbelini, 2021-09-27:
Hi @morrownr I'd be happy to.
Feel free to reach me on my official email: matheus_garbelini@mymail.sutd.edu.sg

Matheus-Garbelini, 2021-09-27:
@morrownr protected management frames (requried ieee80211w=2) at least works well with hostapd.
And in this case, software encryption/decryption is performed correct
[image]
:

morrownr, 2021-09-27:
> Hi @morrownr I'd be happy to.

I will be in touch. I need help.

> This new clean code is here at this repo?
> https://github.com/morrownr/8821au-20210708

That is the new 8821au driver that supports the 8811au and 8821au chipsets. I made that repo public over the weekend and created forwarding links last night as I think it is ready for prime time.

I have created a new repo for the new 8812au driver but it is still private right now. It will be a far more difficult task to get the new 8812au driver ready for prime time. The big problem will be monitor mode support. Monitor mode support is expected on the 8812au chipset and Realtek does a poor job with monitor mode...or at least has in the past so I am expecting a challenge.

morrownr, 2021-09-27:
@Matheus-Garbelini 

I had time this morning so an invitation to the new repo is waiting on you. The new repo is called 8812au-20210629 based on the name of the driver and the date of the code. It is currently in an alpha state. The driver should install and work but it needs work and testing. I'll see you over there.

Nick

closed 2021-09-27 by Matheus-Garbelini