morrownr/8812au #49: Where to find old versions?

FeldrinH, 2024-05-31:
I am looking for an older version of the driver, which, if I recall correctly had monitor mode enabled. I believe the repo used to be https://github.com/morrownr/8812au-20210629, but it now seems to be removed. Is there some way to get this older repo?

luosong0054, 2024-05-31:
Yes. Do you want it?

FeldrinH, 2024-05-31:
Yes, I do want it

morrownr, 2024-05-31:
@FeldrinH 

I had to take the old version down as I have too much work to do. However, for those wanting monitor mode, an effort is underway to create a new driver that is standards compliant and will go into the kernel. I will post an Issue here in this repo to announce when the driver is ready for testing.

FeldrinH, 2024-06-02:
> @FeldrinH
>
> I had to take the old version down as I have too much work to do. However, for those wanting monitor mode, an effort is underway to create a new driver that is standards compliant and [...]

Would it be possible to have the old version up as an archived repo? Even in its current state it was the 8812AU driver with the most reliable monitor mode support out of the ones I have found (at least with my specific setup).

morrownr, 2024-06-02:
@FeldrinH 

> Would it be possible to have the old version up as an archived repo? 

I really don't want to do that. I need to focus my effort on work that will lead to good solutions. I put a message about the old repo going down months before I took it down. If you were using it, then you have it unless you deleted it. 

The new driver can do monitor mode if you go into the Makefiles change the following link as such:

CONFIG_WIFI_MONITOR = y

The monitor mode you get is what Realtek provided. It you find that there is a specific issue that is not working, I might be convinced to bring a specific patch over to this driver but remember, when I am working on this old Realtek driver, I am not working on the new rtw88 driver.

The message I posted in the new driver README is serious:

Note: Monitor mode is not supported. Linux users that want good support for monitor mode in a dual band or tri-band adapter should seek out USB WiFi adapters based on the mt7610u, mt7612u or mt7921au chipsets.

I have adapters based on many chipsets. The usb wifi driver support that Mediatek is providing these days is far superior to anything Realtek is providing, You really should explore your option:

https://github.com/morrownr/USB-WiFi

That is the Main Menu for this site. Reading menu items 1 and 2 can be helpful to Linux users that use usb wifi adapters. Issues is used to post news and for users to asking questions.

Regards,

@morrownr

FeldrinH, 2024-06-03:
> I really don't want to do that. I need to focus my effort on work that will lead to good solutions. I put a message about the old repo going down months before I took it down. If you were using it, then you have it unless you deleted it.

Unfortunately I don't have it, because I installed it using yum from https://aur.archlinux.org/cgit/aur.git/log/?h=rtl8812au-dkms-git and as far as I know yum deletes the source code after compiling. I only looked for the github repo after it was taken down, so I never saw the message about it going down.

> The new driver can do monitor mode if you go into the Makefile. Change the following line as such:
>
> CONFIG_WIFI_MONITOR = y

This seems to work (at least I see some packets in monitor mode), but I currently can't test receiving range. I would really like to have the old driver version (20210629) for comparison.

> It you find that there is a specific issue that is not working, I might be convinced to bring a specific patch over to this driver but remember, when I am working on this old Realtek driver, I am not working on the new rtw88 driver.

I'm not looking for support or maintenance of the old driver. I just want a download of the repository as it existed before going down. The driver that I installed from https://github.com/morrownr/8812au-20210629 through the AUR package has worked with minimal issues so far.

> I have adapters based on many chipsets. The usb wifi driver support that Mediatek is providing these days is far superior to anything Realtek is providing, You really should explore your option:

I am exploring the options and testing multiple adapters, but for now the RLT8812AU adapter that I have is a strong contender in the areas that I care about (channel switching speed and monitor mode receiving range).

morrownr, 2024-06-03:
/home/morrow/Downloads/8812au-20210629-main.zip

Here is a zip file of the old driver. Follow the instructions in the README.

morrownr, 2024-06-03:
[8812au-20210629-main.zip](https://github.com/user-attachments/files/15536397/8812au-20210629-main.zip)

Opps... try number 2. See if this works.

morrownr, 2024-06-03:
> I am exploring the options and testing multiple adapters, but for now the RLT8812AU adapter that I have is a strong contender in the areas that I care about (channel switching speed and monitor mode receiving range).

The 8812au chip is a really good chip. Too bad Realtek usb support is so poor. Hopefully it will not be too long before a good in-kernel driver provided by the community is available.  

> I don't have it, because I installed it using yum 

I know this is not your fault but I have grown to very much dislike the distros that repackage this driver as it ends up causing me problems. If distros are going to repackage a driver, then THEY NEED TO SUPPORT IT! This driver does not need to be repackaged.

FeldrinH, 2024-06-03:
To be clear, AUR contains user provided packages for Arch, so distro maintainers are not to blame. But I agree that packaging experimental drivers like this with no support team to maintain them is problematic.

niraami, 2024-07-04:
Just an FYI, by *deleting* old repos, you can break projects that are already barely maintained, and so will probably never see a fix. Now I have to do an internal fork and add more maintenance work to my already massive pile...
For example, the Yocto Linux layer meta-rtlwifi. It has a recipe to build the driver for 8812au, and uses your repo for it. This is now broken, and only continues to build on machines that have the git repo cached.
See this file for the reference to the repo.
https://github.com/EmbeddedAndroid/meta-rtlwifi/blob/master/recipes-bsp/drivers/rtl8812au.bb

Please archive it next time, I don't understand your arguments against it. Just any disable pulls, and any other notifications that might distract you and users will treat it as a dead/legacy project.
Build machines don't see deprecation warnings. To us devops guys, stuff like this causes massive headaches cause pipelines just *break* one day.

morrownr, 2024-07-05:
@niraami 

I won't charge you for last month's bill. Fair enough?

TRC-jmohedano, 2024-07-18:
Does this version of the driver (20210629) support simultaneous dual-band operation? Thanks in advance!

morrownr, 2024-07-18:
> Does this version of the driver (20210629) support simultaneous dual-band operation?

No

That is not a capability in usb wifi adapters as they only have one radio.